bambulab/BambuStudio #1979: Embedded depth issue when printing lettered signs

CMull87, 2023-07-06:
I 3D print a LOT of signs for work and I’m coming across a “quality” issue that I believe is coming from Bambu studio auto generating an “embedded depth” of 0.2mm in layer height with the letters I place ontop of my solid base of my signs. 

Example: if I’m making a black sign that says “keep out” in white letters on the flat black base. I would model the black base at 1mm tall then place the white “keep out” letters ontop of the black base and make the white letters also 1mm tall then group the pieces together and send to the sclicer.

In prusa sclicer the model mentioned above would print the entire black base first, then would pause for the colour change, then print the white letters ontop of the black base and it would come out nice and smooth.

In Bambu studio the model mentioned above prints the black base but at the 0.8mm tall layer, it starts to outline certain letters in a completely random order (doesn’t outline all letters just some) and then continues until it gets to the 1mm tall layer then pauses changes to white filament and continues printing the letters. 

The “quality” problem I’m having is that when the Bambu studio embeds some of the letters into the black base at 0.8mm layer height (0.2mm into the base layer) it seems to really mess up the smooth layer lines of the base as it’s trying to outline/embed some letters instead of keeping a consistent smooth base until the colour change and letter printing is supposed to begin at layer height 1mm 

I don’t know how to turn off the embedding 0.2mm into the layer below when printing my signs and to be honest the quick and simple way prusa sclicer scliced my files produced a much higher quality print. 

[image]
[image]
[image]

MadMax389, 2023-07-10:
If you set your embed depth to 0, it should print on top of the surface...
[image]

MadMax389, 2023-07-10:
I see what you mean. Even if I place the logo on the surface, it embeds several layers deep. It still prints ok though. 
[image]

CMull87, 2023-07-10:
Yes that’s my experience as well… it prints “ok”…Where I find the issue with the quality I’m used to is the nice smooth consistent layer lines in the “base/background” black piece of the sign. It gets inconsistent and “artefacty” (I just made up my own word there) when the nozzle starts to trace the outline of the letters that the sclicer decides to embed. I just wish there was a way to have the sclicer perform the way prusa sclicer does and just print the base then colour change then print the letters so that way everything is nice a smooth and consistent

CMull87, 2023-07-26:
I don’t get why Bambu Sclicer which is based off Prusa sclicer does this when sclicing a file. Prusa sclicer does NOT embed the depth of letters on a flat surface but Bambu’s sclicer does. I don’t think I can do anything settings wise to correct this (I have tried for hours clicking things on and off and re sclicing with zero success). The alternative would be for me to just use PRUSASCLICER to sclice my files and then send them to my X1C for physical printing, but the activation of the AMS for colour changing is closed source so PRUSASCLICER can not sclice the file with Bambu AMS colour change instructions. 

I just want Bambu sclicer to print my model layer by layer and to place the letters ontop of the base/backing with out AUTO GENERATING an embedded depth of those letters into the previous layers of the base/backing. Why is it doing this? How can I get it to stop and just print layer by layer normally?