bambulab/BambuStudio #1985: Mesh boolean crashes bambu studio

coat64, 2023-07-07:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU AMD 5700G
Ram 64 GB
internal video

### Printer

X1C

### How to reproduce

Make an assembly of two objects and try to do a mesh boolean. When using the buttons, it crashes after clicking on Union, Difference or Intersection. When using the right-click menu it crashes immediately. Doesn't matter if same or different filament.

Computer is restarted after installing the Beta-version

[image]

### Actual results

Bambu Studio Crashes

### Expected results

Not crashing :)

### Project file & Debug log uploads

[BM_crash.zip](https://github.com/bambulab/BambuStudio/files/11984498/BM_crash.zip)
[debug_Fri_Jul_07_19_33_48_11580.log.0.zip](https://github.com/bambulab/BambuStudio/files/11984500/debug_Fri_Jul_07_19_33_48_11580.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MadMax389, 2023-07-08:
This happens for me too...

tubbybtch, 2023-07-09:
I tested for an hour...was able to get *zero* workable booleans.  Crashed BS 6 times completely.  Did this pass alpha?

tangketan, 2023-07-10:
@coat64 Sorry for the bug, but your file is totally fine on my PC. I'm guessing it might be the win 11 causing the issue. Let me check...

@tubbybtch Is your PC also win 11?

tubbybtch, 2023-07-10:
Win 10 Home

coat64, 2023-07-10:
@tangketan No need to apologize, it is a beta version. Maybe of some influence, also running the beta-firmware. I am fine with sharing more information about my system of needed.

tangketan, 2023-07-10:
@coat64 Thanks for your understanding. We have reproduced the crash on another win 11 PC. We'll publish a fixed version ASAP.
@tubbybtch Would you please provide the detailed version info？ Something like this (could copy from system/system info)
Version	Windows 11 Enterprise
Version	22H2
Install date	‎2023/‎7/‎10
OS version	22621.1702
Experience	Windows Feature Experience Pack 1000.22641.1000.0

tubbybtch, 2023-07-10:
***@***.***

On 7/10/2023 5:25 AM, tangketan wrote:

@coat64<https://github.com/coat64> Thanks for your understanding. We have reproduced the crash on another win 11 PC. We'll publish a fixed version ASAP.
@tubbybtch<https://github.com/tubbybtch> Would you please provide the detailed version info？ Something like this (could copy from system/system info)
Version Windows 11 Enterprise
Version 22H2
Install date ‎2023/‎7/‎10
OS version 22621.1702
Experience Windows Feature Experience Pack 1000.22641.1000.0

—

CAUTION This email originated from outside your organization. Please exercise caution when clicking any links or opening attachments.

Brian A. Kamp, MS Senior Lecturer (Retired) Eastern Washington University Dept. of Computer Science

tangketan, 2023-08-10:
The fix is included in Studio V1.7. Please have a try.

closed 2023-08-10 by tangketan