bambulab/BambuStudio #1989: Be able to drop a part of an assembly to the bed

coat64, 2023-07-07:
**Is your feature request related to a problem? Please describe.**
When working with parts in an assembly it is somewhat cumbersome to move a part so it is flat to the bed, especially when you scale it in the Z-direction, a part/modifier/etc laying on the bed is raised from the bed, scaling from the middle of the part.

**Describe the solution you'd like**
A possibility to lay a part on the bed like "lay on face" (rightclick menu?) or a possibility to scale the part relative to the bottom of the part instead of the center.

MadMax389, 2023-07-10:
Second this request. My workaround is to split the assembly to objects so they all lay flat on the bed, then reassemble. Then you can specify the z-height for the object from the bottom surface.
[image]

djngnear, 2024-06-29:
Please add the drop to bed icon like in Prusaslicer!