bambulab/BambuStudio #1994: Wrong hot bed temperature during manual flow rate calibration

sumerland, 2023-07-08:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Additional system information

When running the flow rate calibration the hot bed temperature is set according to the chosen AMS slot / filament. For my generic PLA filament the temperature was 65degC but the printer then set 35degC as target. As a consequence the adhesion was very bad and I stopped the calibration. For PETG the temperature was 80degC but the printer never set a target temperature after downloading and unzipping the model.

### Printer

Bambu Lab P1P

### How to reproduce

Calibration -> Flow Rate -> Manual calibration -> select filament and start calibration.

### Actual results

Hotbed target temperature on printer is different from the temperature set in the software.

### Expected results

Hotbed should be heated up to target temperature from the slicer.

### Project file & Debug log uploads

[debug_Sat_Jul_08_11_43_31_35430.log.0.zip](https://github.com/bambulab/BambuStudio/files/11990384/debug_Sat_Jul_08_11_43_31_35430.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

serkanaksu, 2023-07-09:
Can confirm, its using only Cool Plate settings.

CoryCharlton, 2023-07-10:
Was just about to create an issue on this. I have selected the textured PEI plate and the temperature box indicates 55C (which is in itself not the temperature I want to use and cannot be changed) but the process doesn't actually heat the bed to more than 35C. Manually adjusting the temperature appears to work but it does not get to full temperature before the test starts so there are issues with the first layer.

sumerland, 2023-07-13:
Solved in v1.7.0 Beta2 - Thank you

closed 2023-07-13 by sumerland

zen, 2023-10-27:
I still see the same issue in 1.7.7.89 on MacOS. 
Last screen shows 35 degrees and bed never goes above that. All calibration attempts fails due to very bad adhesion.