bambulab/BambuStudio #1995: v1.7.0 Public Beta, Windows 10, Login impossible Blank login screen

rainbowat, 2023-07-08:
### Bambu Studio Version

1.7.0

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab X1

### How to reproduce

I just clicked login and Blank login screen

### Actual results

Blank login screen

### Expected results

login screen

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-07-09:
This maybe a network problem. Please try again some time later.

seler, 2023-07-20:
I have exactly the same issue in v1.7.0 Public Beta 2. Also, it does not see my printer in LAN (without login).

rainbowat, 2023-07-20:
After two weeks, no change
It still doesn't work for me.

rainbowat, 2023-07-20:
> @rainbowat I was able to overcome the problem by downgrading to 1.6, logging in there and then upgrading back to 1.7.

I just started using orca-slicer

rnunez24, 2023-08-04:
I'have same isue 1.7.1.62 version.

bambu123, 2023-08-04:
> I'have same isue 1.7.1.62 version.

Please send me the log files.
Remove cache directory %LOCALAPPDATA%\BambuStudio\EBWebView and retry.

rnunez24, 2023-08-04:
[debug_Fri_Aug_04_11_40_42_3712.log.0.txt](https://github.com/bambulab/BambuStudio/files/12259890/debug_Fri_Aug_04_11_40_42_3712.log.0.txt)
[image]

Cache deleted and retryed. 
Uninstalled and reinstalled tried

rnunez24, 2023-08-04:
Solved: in preparation view an advertisement of network plugin appears. 

Installed & now the login works.

[image]

arnrist, 2023-08-06:
I experienced the same issue, but different resolution. Placing a comment here in case there is something to do in the application - or other have the same issue.
In my case the network plugin were installed.
First install on a windows 10 PC.
Version 1.7.2.51.

In my case the "Language" was not selected in the General Settings.
After I went to Preferences and selected the language, I was immediately able to get the login screen.

I had tried multiple settings before this, including reinstalling the application, checking the firewall settings, verifying the Network plugin etc.
The language setting fixed it for me.

bambu123, 2023-08-07:
> I experienced the same issue, but different resolution. Placing a comment here in case there is something to do in the application - or other have the same issue. In my case the network plugin were [...]

@arnrist Thank you for your solution. To fix this problem totally, could you please send me your logs for analyzing?

bambu123, 2023-12-21:
Fixed in v1.8.2.

closed 2023-12-21 by bambu123