bambulab/BambuStudio #1996: Remember open directory when saving project

cxen, 2023-07-08:
**Is your feature request related to a problem? Please describe.**
BambuStudio does not seem to remember the folder directory from which an .stl was opened. This is annoying when saving a project, because the save dialogue opens at the last saved directory, often a completely different project.

**Describe the solution you'd like**
I would like BambuStudio to remember the directory from which an .stl has been opened and offer that directory in the save project dialogue. PrusaSlicer does this already.

bambu123, 2023-07-10:
This has been fixed in the new 1.7 beta version. Are you using the new version.

cxen, 2023-07-10:
> This has been fixed in the new 1.7 beta version. Are you using the new version?

I just tried the 1.7 beta and it works, thank you

closed 2023-07-11 by bambu123