bambulab/BambuStudio #1997: Add GCode viewer / progress screen

AdamLeyshon, 2023-07-08:
**Feature Request**
Please add a view which allows you to see the current print job as visualised gcode,
Ideally this would be synchronised with the printer so we can track exactly where it is in the job.

This is a feature I miss a lot from OctoPrint

When you have very large and complex jobs, looking at the current layer number isn't entirely useful and if you've cleared the build plate, the preview screen cannot help you find where it is in the current print.

The GCode viewer is also useful for finding problematic areas that are causing a print to fail or look wrong, possibly due to excessive number of retractions for example.

[image]

Since we cannot integrate third party software at present, it seemed reasonable to ask for the feature.

RJSavell, 2023-10-08:
I was thinking the same thing - Ye,s please add this as an option when viewing the live print.

MarkDerbyshire, 2023-11-19:
This would be a great addition.  I miss the Octoprint viewer

p1ranha, 2023-11-25:
This would be really really handy considering slow preview of camera on A1 Mini. It should be possible to switch from camera view to gcode current layer preview.

WolfspiritM, 2023-12-15:
Would love this, too. As a first step it would even be enough for me to have a small version of the "Preview" tab visible in the device tab with everything up to the current layer visible. Maybe rotated per default to where the camera would be. I don't really need the steps inside a layer. But it would help to have a visual comparison between how it looks and how it should look.

tobiberger, 2024-01-25:
This is definitely one of the main things I'm missing since switching to a Bambulab printer, coming from an Octoprint setup. Hope this gets picked up soon.

nex6, 2024-04-11:
Full agreement with the rest of you, love my Bambu but the live G-code viewer is something I am really missing

MartynWilliams, 2024-05-14:
Bump ... This would be a great feature. Alas no one as been assigned to look at the suggestion.

ProtoBotic, 2024-05-27:
i like the gcode plot/veiwer below.
https://github.com/bambulab/BambuStudio/assets/155253484/dac4f1e7-a366-4260-96b5-30948f474f7f

OGJakearroo, 2025-01-17:
+1 for this one.
I find 3d G-Code viewers far nicer than the basic layer by layer viewers, but something more than x/x layers would be nice.
Also estimated finish time doesnt work which is weird.

keprice20, 2025-03-15:
Same here, just looking for a third-party feature to see if it was available.