bambulab/BambuStudio #2003: Add Orcaslicer to "Based on" text in README.md

Apulo, 2023-07-08:
### Bambu Studio Version

01.07.00.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 Series/P1P

### How to reproduce

Read README.md on release page

### Actual results

Bambu Studio is based on [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) by Prusa Research, which is from [Slic3r](https://github.com/Slic3r/Slic3r) by Alessandro Ranellucci and the RepRap community.

### Expected results

This text should include Orcaslicer since many Orcaslicer features have been incuded in Bambu Studio in this release

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

craysiii, 2023-07-09:
OrcaSlicer is itself a fork of BambuStudio, this makes no sense. Sure they can say that they have pulled in changes from OrcaSlicer, but to claim that OrcaSlicer was first is silly.

Apulo, 2023-07-09:
My wording may have been off here. There should be some additional attribution provided for features merged from Orcaslicer. Maybe "Includes additional features from Orcaslicer" or something similar.

jphoke, 2023-07-10:
It is fairly customary to note that features were pulled back into the project from a fork ... not "based on" ...either way it should be included IMO

Photogad, 2023-07-11:
All the complaint about BambuStudio pulling "all of OrcaSlicer features" recently make no sense to me... Bambu Studio still lake many things OrcaSlicer have, such as all the calibrations, the ability to set pressure advance, etc. No offense to SoftFever but to say Bambu took all of his code is a bit wrong since OrcaSlicer still has a lot more settings and is better

closed 2024-01-16 by SaltWei