bambulab/BambuStudio #2004: Cannot select Filament for Calibration in Flow Dynamics Calibration

robin7331, 2023-07-09:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Additional system information

CPU: Apple M1 Pro
RAM: 32 GB

### Printer

Bambu Labs P1P

### How to reproduce

1. Open Bambu Studio
2. Select Tag "Calibration"
3. Select Printer
4. Select "Flow Dynamics"
5. Choose Nozzle
6. Choose Print Plate
7. Choose Filament <- BUG

### Actual results

Cannot choose any filament
(AMS is not installed)
[image]

### Expected results

I would Expect to see a list of filaments to choose from.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

boswelja, 2023-07-12:
Same issue on Windows 11, it doesn't seem to be exclusive to macOS

LiZ-Li-BBL, 2023-07-17:
We fixed this issue on new release [v1.7.0 Public Beta 2](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.65)

closed 2023-07-17 by LiZ-Li-BBL