bambulab/BambuStudio #2007: "Flush into this object" (and other flush options) crashes Bambu Studio

tangcla, 2023-07-09:
### Bambu Studio Version

1.7

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. Load two objects
2. Set one of the objects to a different colour
3. Right-click one of the objects --> Flush options --> flush into this object OR flush into objects' infill OR flush into objects' support
[image]

### Actual results

Mouse cursor turns into hourglass, program crashes

### Expected results

I expected my flush option to be enabled.
I also was unable to find another way to 'flush into this object' in the left-hand pane menu, despite me using the search and it showing as an option.
[image]

### Project file & Debug log uploads

[flush_test.zip](https://github.com/bambulab/BambuStudio/files/11994975/flush_test.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/11994977/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-07-10:
Thanks! We will fix asap.

jphoke, 2023-07-10:
Confirming this also happens on macOS 13.4.1 and current beta Studio

LiZ-Li-BBL, 2023-07-17:
We fixed this issue on v1.7.0 Public Beta 2 (https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.65)

closed 2023-07-21 by LiZ-Li-BBL