bambulab/BambuStudio #2008: Tree supports no longer have interface layers on top

RyanEwen, 2023-07-09:
### Bambu Studio Version

1.7.0 public beta

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

Use manual tree supports on a flat overhang. 

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[85mm.zip](https://github.com/bambulab/BambuStudio/files/11995444/85mm.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DokuKaefer, 2023-07-09:
If you deactivate the option "On Build Plate only" , then it works again.

RyanEwen, 2023-07-09:
> DokuKaefer

Can confirm, this does work around the issue. Thanks

jphoke, 2023-07-10:
Workaround is viable on macOS version as well, but is craptacular for very complex models that I do not want supports other than on the build plate :(

tangketan, 2023-07-12:
Thanks for reporting the issue. We have a logic to add supports at corners for tree(auto), now I extend it to tree(manual) also. Do you think it's OK?
[image]

RyanEwen, 2023-07-12:
> Thanks for reporting the issue. We have a logic to add supports at corners for tree(auto), now I extend it to tree(manual) also. Do you think it's OK? [image]

I like this idea a lot and was hoping something would be developed for it one day. It seems unrelated to the issue here, though.

tangketan, 2023-07-12:
@RyanEwen It is related in a very complicated way... 
1) this overhang area is too small, so normally only one tree node is generated.
2) if "On Build Plate only" is not turned on, we can merge the original overhang to the support interface, that's the workaround proposed by @DokuKaefer . We can't do this when "On Build Plate only" is on, because in some cases we may get dangling support interfaces.
3) a better way to solve it is: adding additional tree nodes at the corners of the support. Previously we only do this for auto-detected overhangs, now I extend it to manually painted overhangs too.

RyanEwen, 2023-07-12:
That makes sense. I had been working around the issue by reducing the `Tree support branch distance` to 1mm, which worked nice with the extra interface layers (until they stopped appearing due to my use of `On build plate only`

[image]

tangketan, 2023-07-19:
@RyanEwen Yes, reducing the _Tree support branch distance_ is a universal solution. My method only works on surfaces that have sharp corners. If the surfaces are round, it won't work.

hellojoewong, 2025-01-09:
good job