bambulab/BambuStudio #2009: Calibration Not Working for PETG

mltriebe, 2023-07-09:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Select manual calibration
2.  Select high temp plate
3. 70 degrees for bed temp is the default
4. PETG for the filament.

### Actual results

When it goes to print the pattern the build plate is 0 degrees. It did this several times.

### Expected results

The build plate should heat to 70 degrees.

### Project file & Debug log uploads

[debug_network_Sun_Jul_09_09_00_09.zip](https://github.com/bambulab/BambuStudio/files/11995751/debug_network_Sun_Jul_09_09_00_09.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bochom, 2023-07-11:
Instead of heating bed software heats nozzle.

DokuKaefer, 2023-07-11:
I have the same problem with PLA and the textured PEI plate. 
The plate at the test print has only 35 degrees instead of 55 degrees as indicated.

zhimin-zeng-bambulab, 2023-07-17:
the same with https://github.com/bambulab/BambuStudio/issues/1994
It has been solved in v1.7.0 Beta2

closed 2023-08-04 by zhimin-zeng-bambulab