bambulab/BambuStudio #2010: Can you please add Paint on Fuzzy Skin.  Another great feature would be have multiple paint on textures. Thanks

SeaStorm00, 2023-07-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Notarin, 2023-07-11:
This is a duplicate of #1938

Notarin, 2023-07-26:
perhaps migrate the setting of it being a toggle to a new painting tool, similar to the color painter, it could be a texture painter, this would be scalable as well.
smooth, zigzag, fuzzy, pointy, etc.