bambulab/BambuStudio #2012: Strange behavior with the tree support

DokuKaefer, 2023-07-10:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Load the following **_Backing peg.stl_** to a new project:
https://www.printables.com/de/model/119811-skadis-universal-peg/files

Choose option "On build plate only"

### Actual results

As you can see on the screenshot, the support is not applied to the object facing forward.

[image]

As soon as the option "On build plate only" is deactivated. The support is also added to the front object.
But still different to the other ones.

[image]

With Bambu Studio version 1.6.2 it is even worse.

[image]

### Expected results

The tree support should be correctly applied to all objects in the same way.

### Project file & Debug log uploads

https://www.printables.com/de/model/119811-skadis-universal-peg/files

Backing peg.stl

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DokuKaefer, 2023-07-10:
Another example. **_I know the printer can do this without support. I just want to point out the bug here._**
https://www.printables.com/de/model/121329-skadis-screwdriver-rack/files

Bambu Studio 1.7.0.53

One is ok, the other same hole is without support.
[image]

The back would be better for support.
[image]

And if you change the orientation on the plate, will it get better with the support, but now the outer holes would be missing.
[image]

In this orientation the outer holes would be missing again and the third hole from the left.
[image]

The best orientation is diagonal.
[image]

RyanEwen, 2023-07-11:
Sounds possibly related to https://github.com/bambulab/BambuStudio/issues/2008

tangketan, 2023-07-12:
The generation of tree nodes is discrete and depends on orientation. It is not isotropic. I'll need a little more time on this...

tangketan, 2023-07-12:
@DokuKaefer Automatically solving this issue is difficult, but you can manually set a smaller branch distance
[image]

PowerChaos, 2023-07-12:
i going add it to this post here

Orca Slicer 1.6
[image]

Bambu Studio 1.7
[image]

so overhangs of 85 degree get ignored 
and it does overhangs on places it is not needed

hopely this can help

Greetins from PowerChaos

tangketan, 2023-07-20:
@PowerChaos Would you please share the 3mf?

PowerChaos, 2023-07-20:
[Bunny Full.zip](https://github.com/bambulab/BambuStudio/files/12101605/Bunny.Full.zip)

> @PowerChaos Would you please share the 3mf?

i attached the file , no idea where i got it from 
also this version is the assembled version , i got also the unassembled version in 11 parts
in the end the print was succesfull as it does not need support at all ( but benefit from it under the ears/chin ) 

Eather way , same settings and so much difference , so somehting is wrong with the calculation of hte supports xD

tangketan, 2023-08-02:
Bambu Studio 1.7 has just been released, and the default tree supports are good enough to me.
[image]

DokuKaefer, 2023-08-03:
@tangketan i have now the final version 1.7.1 and was about to print the following model:
https://www.printables.com/de/model/433736-govee-temphumidity-sensor-model-h5179

Version 1.7.1
[image]

Version 1.6.2
[image]

tangketan, 2023-08-10:
This is a bug with small overhang removal. I'll look into it.
For now you can disable "remove small overhangs" option to workaround.
[image]