bambulab/BambuStudio #2014: Flow calibration checkbox not retained between prints

MadMax389, 2023-07-10:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

10th gen Core i5
32 GB
RTX3060

### Printer

X1C

### How to reproduce

1. Send any print to the printer.
2. Untick Flow Dynamics Calibration checkbox
3. Send print again, observe status of checkbox

### Actual results

Checkbox status is not retained between prints
[image]

### Expected results

This behavior was fixed in v 1.6 but seems to have made its way back in 1.7. 
Item 8 in 1.6 release notes _"Retain the most recent print settings for bed leveling, flow calibration, and timelapse to streamline the user experience https://github.com/bambulab/BambuStudio/issues/487 https://github.com/bambulab/BambuStudio/issues/1035 https://github.com/bambulab/BambuStudio/issues/1081"_

### Project file & Debug log uploads

None required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-07-14:
Hi. What type of bed do you have PTE?

MadMax389, 2023-07-14:
I usually use a 3rd party plate with plate detection off. I have modified my PLA filament profile to to use 60C bed temp on all plates so I don't need to think about what plate is selected (except for textured plate which lowers the nozzle a little bit). I normally leave cool plate selected.
[image]

walterwongbbl, 2023-07-18:
Sorry, this issue does exist and we will fix it

closed 2023-07-31 by lanewei120

walterwongbbl, 2023-08-02:
Please update the latest studio version