bambulab/BambuStudio #2015: PLA layers squished following PETG support interface layers

RyanEwen, 2023-07-11:
### Bambu Studio Version

1.6.2, 1.7.0 public beta

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

Use settings like so:

[image]

Print a model with a flat overhang needing supports (I use normal / manual with the overhang painted using the fill tool)

[image]

[image]

### Actual results

Note the bulged layers all the way around the entire model, aligned with the PETG intefacing

[image]

I have tried using 0.2 mm Top Z distance as well, in case that was the issue. It did not help.

### Expected results

Expected the PLA portion of the model to appear how it does in the slicer preview, or when using PLA supports.

### Project file & Debug log uploads

I cannot share this particular project unfortunately.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-07-11:
Not slicer issue. Can't see the line on preview page.
Could you share gcode.3mf (no model 3mf) for us to reprinting?

SaltWei, 2023-07-11:
> 1.6.2, 1.7.0 public beta

@bambu123 Not issue related 1.7.0 version

RyanEwen, 2023-07-11:
> Not slicer issue. Can't see the line on preview page. Could you share gcode.3mf (no model 3mf) for us to reprinting?

Yes it does look okay in the sliced preview. 

FWIW I have tried with various filaments including Bambu PLA. I have printed many different variations of the model as well, but it always happens. I used to have a second supported structure at a different height and the same bulge happened there as well. 

Attached gcode is uses Bambu Basic PLA in AMS slot 1, Bambu Basic PETG in AMS slot 2. 
The file with CUT in the name is a shorter / cut up version of the same model, to save you time and filament. 

[gcode.zip](https://github.com/bambulab/BambuStudio/files/12009894/gcode.zip)

SaltWei, 2023-07-11:
OK. We will reprint this gcode file and try to find and analyse the cause.  Thank you very much!
@QingZhangBambu Please follow this issue.

Hologos, 2023-07-11:
What is your flushing volume between the PLA and PETG? You have to flush at least 700 to the nozzle is clean. PLA and PETG does not stick to one another so if the nozzle is not clean, PLA and PETG will mix and cause a layer separation.

RyanEwen, 2023-07-11:
> What is your flushing volume between the PLA and PETG? You have to flush at least 700 for the nozzle to be clean. PLA and PETG does not stick to one another so if the nozzle is not clean, PLA and PETG will mix and cause a layer separation.

I use 800, but the gcode attached above might be only 250 as I rearranged the filaments (moved PETG to slot 2 and deleted 3).

I will reslice and attach new gcode example.

Some other slicer settings I should mention that I'm using:

[image]
[image]

RyanEwen, 2023-07-11:
It occurred to me that I'm using 4 interface layers while the dropdown only offers the choice between 1, 2, or 3 (I manually typed in 4). I thought maybe that could be causing the issue so today I printed with just 2 interface layers instead of 4 (I didn't choose 3 because going down to 2 saves a couple of filament changes). Unfortunately the issue remains. 

I've also updated the photo in the OP with a better one.

QingZhangBambu, 2023-07-12:
We will check this.

Hologos, 2023-07-12:
@RyanEwen This happens to me as well when there is a transition from sparse infill to solid layers. Isn't this the case?

RyanEwen, 2023-07-12:
@Hologos that's a different issue but coincidentally that also happens in this model. You can see what you described just above my thumb in the photo in my original post.

The bulge I'm trying to resolve is line with the white PETG support interface layers and there's no change in infill in the model there. 

In a previous version of this model there was post that stuck out of the diamond shaped hole, and that post required supports. The same bulge occurred there as well.

Hologos, 2023-07-12:
@RyanEwen I downloaded the gcode you posted and it's kinda weird. There is no transition near the outside wall with the first ring flaw, there is a transition near the inner wall. I don't see anything that should cause the second one. Is the ring all the way around? It kinda looks like a layer shift (the upper part is moved forward).

RyanEwen, 2023-07-12:
@Hologos the ring aligned with the PETG support interface goes around the entire print, sticking out equally on all sides. It sticks out a bunch on one layer, then less on the next, and less on the next, until back to normal agian.

It's almost as if it printed 2 layers at the same Z height, but the print preview shows otherwise.

RyanEwen, 2023-07-12:
I should also mention that the ring sticks out on the inside walls and within the holes as well. Not just the exterior of the print.

[image]

abc77654, 2023-07-15:
[image]
[image]
[image]
即使普通支撑打印也会有支撑面的错误挤压线

Hologos, 2023-07-15:
Is the support layer height the same as normal layer height?

RyanEwen, 2023-07-17:
[image]
[image]

@GurselGokce interesting observation with the layer time - but it's simply due to the PETG interfacing adding a lot of time to those layers.

I just printed this model again today and I was able to stand by and watch as things happened. What I noticed is that the 2nd layer of PETG interface didn't seem like it was 0.20mm high. It appeared less to my naked eye - almost like ironing. The PLA portion afterwards was similar - it didn't look like it was 0.20mm higher than the previous layer.

GurselGokce, 2023-07-17:
@RyanEwen The PLA parts printed at the same layers as the PETG support interfaces are also at a lower fanspeed than the rest of the PLA layers. I've tested it just now and it lowers my part fan to 40% while printing the PLA part of the layer before AMS switches over to the PETG interface to finish printing that layer. This happened for 3-4 layers now, which is a consistent behavior I'm guessing.

[image]

No idea how much printing the PLA at a lower fan speed would affect the print as I'm fairly new to this stuff, but I wanted to bring it up as well.

RyanEwen, 2023-07-17:
@SaltWei @QingZhangBambu should I open a support ticket with a log attached in case this is firmware related? Or will you communicate with firmware team yourselves after confirming that this isn't caused by the slicer?

RyanEwen, 2023-07-25:
@SaltWei @QingZhangBambu I have created a support ticket in case the issue is firmware. Ticket # US230725827003

T00LF00L, 2023-07-28:
So two things..
1. You only need one interface layer.
2. This is just an unfortunate result of basically pausing a print to switch material and temperature to print something else, then switching material & temperature back to resume your print. One interface layer means you’ll only do it once and the effect will be minimized.

Hologos, 2023-07-28:
> So two things..
>
> 1. You only need one interface layer.
>
> 2. This is just an unfortunate result of basically pausing a print to switch material and temperature to print something else, then switching [...]

I've done this method several times in the past and have never came across what happened to these guys. I don't have a PETG on hand but when I get some, I'll test some old project that didn't produce this to test if it's firmware (since the gcode wll be the same).

RyanEwen, 2023-07-28:
@T00LF00L 

> 1. You only need one interface layer.

The reason for 4 interface layers is that I get a _significantly_ better finish on the supported part. I've done a lot of testing regarding this: https://github.com/bambulab/BambuStudio/issues/2095

> 2. This is just an unfortunate result of basically pausing a print to switch material and temperature to print something else, then switching material & temperature back to resume your print. One interface layer means you’ll only do it once and the effect will be minimized.

It's actually just 1 filament change per 2 interface layers since the slicer is smart enough to switch to PETG at the end of the layer and start the next layer PETG before switching back to PLA.

If it was temperature related then I don't think I would see the 2nd PETG interface layer so squished as it prints, which happens without any pause or material change. I see 2 PETG layers print back to back, the 2nd one appearing to be hardly any higher in terms of Z height than the first (almost looks like ironing it's so squished).

QingZhangBambu, 2023-07-31:
@RyanEwen I reproduced the problem and found that the bottom of the model is not level. I presume that the printing plate got warped during printing. I will do some tests to confirm that.
[image]

QingZhangBambu, 2023-07-31:
@RyanEwen and if you don't mind, could you provide 3mf file to help me do some tests, not just gcode.

RyanEwen, 2023-07-31:
@QingZhangBambu are you able to access the 3mf if I attach it to my support ticket (for privacy)?

RyanEwen, 2023-07-31:
[Bambu Support Project - Github issue 2015.zip](https://github.com/bambulab/BambuStudio/files/12215249/Bambu.Support.Project.-.Github.issue.2015.zip)

QingZhangBambu, 2023-08-01:
@RyanEwen Thanks for the file, and can you put your model on the table and take a picture like mine. I want to take a look to help find the cause.

RyanEwen, 2023-08-01:
I've printed over 50 of these now. Always nice flat bottoms and tops except rare instances where filament didn't stick well to the bed.

[image]

QingZhangBambu, 2023-08-01:
thanks, dude. I will find the reason as fast as possible.

QingZhangBambu, 2023-08-02:
@RyanEwen This has to do with layer time. Prolonged cooling will cause the lower layer to shrink inward. causing the contact layer to protrude. One side of the fan protrudes more than the other.  Based on this speculation, I changed the support interface material to PLA, and the layer time was reduced from 259.7s to 200s, and the result was improved.
the upper used PETG as interface, the lower used PLA
[image]

RyanEwen, 2023-08-02:
@QingZhangBambu can you reproduce the issue by adding a 59.7s pause when using only PLA?

QingZhangBambu, 2023-08-15:
@RyanEwen I do some tests, and you are right, layer cooling is not the only reason. I may need more time to find the cause. Sorry for the delay.

attilah, 2023-08-27:
I have a really similar issue for this part, I print it in PLA with Bambulab Support W, and it separates at 2 places: when the sparse infill changes to the wall part (the lower colored parts), and where the support material is. This is the layer speed graph on the pic.
[image]

abc77654, 2023-08-30:
[image]
[image]
这个问题层线我发现是因为支撑层顶面有拉桥，导致打印速度降低，打印时间增大，模型的外墙的冷却时间更长收缩了，由于打印默认参数顺序是：内墙—外墙—填充，所以等到打外墙时候，外墙已经相比基础层的外墙冷却过长，再继续打就发生的断层线，所以通过更改打印顺序，改成外墙—内墙——填充可以解决。
[image]

（deepl翻译：This problem layer line I found because the top surface of the support layer has a drawbridge, resulting in a reduction in printing speed, printing time increases, the cooling time of the model's exterior wall is longer contraction, due to printing the default parameter order is: interior wall - exterior wall - infill, so wait until you hit the exterior wall when the exterior wall has been compared to the foundation layer of the exterior wall cooling is too long, and then continue to hit it on the The fault line occurs, so by changing the print order to outer wall - inner wall - infill can be solved.）

attilah, 2023-08-30:
@abc77654 I wish this would be a slicer feature to detect such walls. I ended up with not using support material and add thicker walls (to have infill) and arachne wall generator and these made both problems go away...

mardarelaurentiu, 2024-01-31:
Just had the same problem. The gray filament is PLA, the white one is PETG interface support. 
Printed with top cover and door closed. aux and chamber fan off. I have those lines at every print with petg interface support. Didn’t try with other support interface materials.

RyanEwen, 2024-01-31:
> @RyanEwen I do some tests, and you are right, layer cooling is not the only reason. I may need more time to find the cause. Sorry for the delay.

Did you ever find out what else is contributing to this? It does seem that the layer time plays a large part of it for sure, so I've worked around that in my project. 

@mardarelaurentiu it appears that your image wasn't finished uploading when you submitted your message

dg4prez, 2024-01-31:
This definitely happens with Bambu pla support material as well. You can see it in the model and the prime tower, even at 900 purge.

gaolegao-lx, 2024-08-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

brinkabeler, 2024-08-28:
I have the same issue while printing PETG with PLA support. There is a PLA layer within the whole print, not just where the interface should be.

RyanEwen, 2024-08-28:
That's more likely due to insufficient flushing. Try setting it to 800 in the flushing table whenever changing to it from PLA.

Hologos, 2024-08-28:
> I have the same issue while printing PETG with PLA support. There is a PLA layer within the whole print, not just where the interface should be. 

You have to be more careful with flushing now as the slicer now recalculates flushing volume automatically whenever you change color or add new filament. It can be disabled in the settings.

brinkabeler, 2024-08-29:
> You have to be more careful with flushing now as the slicer now recalculates flushing volume automatically whenever you change color or add new filament. It can be disabled in the settings.

You’re correct. I was trown off by an earlier comment saying this was not the problem. Thank you.