bambulab/BambuStudio #2018: ?????? ?????? replacing text

kolomiecalexa2, 2023-07-11:
### Bambu Studio Version

1.7

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Additional system information

hello, I have a problem with language files. when installing the Russian language and when slicing instead of text ?????? can you tell me how this can be fixed ? and when will you be able to change the Russian language to the standard settings?(I have attached the translation file if you can add the Russian language, please, without it in any way)

### Printer

bambu Lab X1 Carbon

### How to reproduce

after replacing the language file in the Bambu_Studio_win-v01.07.00.53-20230707211436\resources\i18n folder, everything was transferred perfectly, but in the tips and upper buttons instead of text ?????

### Actual results

[image]

### Expected results

russian text

### Project file & Debug log uploads

https://github.com/SoftFever/OrcaSlicer/issues/1462  link to the translation file

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-07-13:
Hello, @kolomiecalexa2 

this is just because the Russian language selection missing, which leads to the fonts of Russion not loaded by imgui

we are considering to add Russian support next version

could you update your pot file based on our newest codes and submit it to github?

LiZ-Li-BBL, 2023-07-13:
You need to select the correct language from the preferences settings so that part of the text which appears as "?" will be loaded properly. Unfortunately this is not working at present.

We plan to add Russian to preferences settings in the future.

kolomiecalexa2, 2023-07-17:
[BambuStudio_ru.zip](https://github.com/bambulab/BambuStudio/files/12067811/BambuStudio_ru.zip)
g

translated using Poedit by Google Machine translation.

iStarComLab, 2023-07-24:
та же тема https://github.com/SoftFever/OrcaSlicer/issues/757

iStarComLab, 2023-07-24:
> You need to select the correct language from the preferences settings so that part of the text which appears as "?" will be loaded properly. Unfortunately this is not working at present.
> 
> We plan to add Russian to preferences settings in the future.

*****************************************************************
Добрый день, когда ждать данное решение?

iStarComLab, 2023-08-06:
РЕШЕНО!

LiZ-Li-BBL, 2024-01-16:
We have added support for several new languages including Russian.

closed 2024-01-16 by LiZ-Li-BBL