bambulab/BambuStudio #2019: [BUG] Unwanted supports even if SUPPORT BLOCKER is added.

bolsoncerrado, 2023-07-11:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura

### Printer

X1C

### How to reproduce

Just adding support to a tiny area, I try to enforce block of supports but it wil still create silly supports.
Using manual supports as follows:

[image]

[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

I can't share the file but it should be easy to re-create/reproduce on your end.

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

jwilliams1976, 2023-07-11:
This issue exists in v1.6.3 and public beta 1.7 as well.

tangketan, 2023-07-12:
This is a long-existed issue. The blocker only prevents new supports generated, but won't prevent upper supports grow into the area. I'll try fix it.

tangketan, 2023-10-24:
@bolsoncerrado 3mf file, please?

bolsoncerrado, 2023-10-24:
Wow it's a july issue I'm not sure if I have that file anymore but it's related to the other issue I opened so I guess everything connects...