bambulab/BambuStudio #2020: Impossible to use Beta 1.7.0.53

foresta86, 2023-07-11:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Impossible to use the new features of this beta because it says printer is not updated to latest version but there is no newer version than 01.05.02.00

### Actual results

[image]

### Expected results

A newer firmware

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12015561/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

intdel, 2023-07-11:
The firmware is still in beta.

foresta86, 2023-07-11:
> The firmware is still in beta.

And is there a way to download it as I did for Bambu Studio beta?

intdel, 2023-07-11:
No the sign up is already closed:

https://forum.bambulab.com/t/opportunity-to-join-our-x1-firmware-beta-test/8134

closed 2023-07-11 by foresta86