bambulab/BambuStudio #2021: Remove misleading picture from wiki regarding calibration

intdel, 2023-07-11:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1C

### How to reproduce

1. Go to https://wiki.bambulab.com/en/software/bambu-studio/calibration_pa
2. Use (human) eyes to look at picture in step (3) Calibration printing in progress.
3. Use (human) eyes and brain to recognize textured PEI plate in picture

### Actual results

Picture of textured PEI sheet.

### Expected results

Should not show textured PEI plate, because it is not compatible, as mentioned earlier in the wiki.

### Project file & Debug log uploads

[https://wiki.bambulab.com/software/bambu-studio/calibration/pa_auto_cali.jpg](url)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-07-12:
[image]

zhimin-zeng-bambulab, 2023-07-20:
@intdel  Thank you very much for pointing out this issue. We have updated the image on the wiki and will update the software image in the next version.

zhimin-zeng-bambulab, 2023-08-18:
It has been fixed in the latest version.

closed 2023-08-18 by zhimin-zeng-bambulab