bambulab/BambuStudio #2029: Add a 5th selectable print speed setting

bl4ckm0onk3y, 2023-07-12:
**Is your feature request related to a problem? Please describe.**
The P1P has 4 speed modes, selectable on the LCD panel on the device, or in handy app/bambu studio app.
Sometimes the 100% aka "normal" speed is too much, and the 50% "silent" is too slow, would be nice to have another intermediate speed like 70-75% to select, maybe consider adding it by having 
-silent 50%
-quality 70-75% (because i know you don't want to make slower speed a norm, because of the marketing ect. 
-normal 100%
-sport 124%
-ludicrous 166%  

**Describe the solution you'd like**
-add a "quality" option for print speed with 70-75% (for PLA SILK filaments or other materials that need that speed reduction) 

**Describe alternatives you've considered**
Alternative could be adding a "%" of the speed as a type in number from the app/bambu  studio before print ?

Notarin, 2023-07-14:
+1 on setting speed by number input, why is a this not available? every other printer I've ever used has this available

Dilbertus, 2024-03-28:
Nothing changed here in month? 
Why not set direct percentage like others? A setting between 100% and 50% is a must. 50 % does slow down to much, but anything about 75% does improve quality significantly.
Any naming it "Quality" will surely affect any review, because then it will compare better to competition...

Roemer, 2024-08-20:
I would also absolutely love a 75% speed setting. Like, I want it a bit slower but not 50% slow.

darrepac, 2025-02-06:
Sounds like an easy to implement feature request and a must: between 50% and 100% the difference is too big, a 75% makes sense at least

aimbdd, 2025-08-29:
Please Bambu? I hate that its either 50 or 100%.

Gask44, 2026-09-05:
I would also really like to see a 70–75% speed mode added between Silent (50%) and Standard (100%).

This would be especially useful when printing ASA, ABS and other engineering filaments.

For example, I am currently printing a large multi-part ASA project on a Bambu Lab H2S. At 100% speed I started seeing some corner warping, so I switched to Silent mode at 50% to give the print more stability. It helped, but the jump from 100% to 50% is simply too large — the remaining print time increased dramatically.

A 75% mode would be a very useful middle ground: slower and gentler than Standard mode when extra print stability is needed, without almost doubling the print time by switching to Silent mode.

Something like:

Silent — 50%  
Stable / Quality — 75%  
Standard — 100%  
Sport — 124%  
Ludicrous — 166%

This would be particularly valuable for ASA, ABS, PC and other engineering materials, as well as large prints where reducing speed can help once signs of warping or instability appear during an ongoing print.

It would be even better if 75% could be selected directly from the printer, Bambu Studio and Bambu Handy just like the existing speed modes.

bl4ckm0onk3y, 2026-09-06:
3 years later still nothing.

Roemer, 2026-10-17:
Still nothing unfortunately.