bambulab/BambuStudio #2037: Distorted text in 3D view with 1.7 beta on Linux

mzanetti, 2023-07-13:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.04

### Additional system information

CPU: Intel 10th gen i9
GPU: Intel 10th gen i9 and/or nvidia 

### Printer

X1C with AMS

### How to reproduce

Start the 1.7 beta on Ubuntu. Import a STL file.

### Actual results

See fonts in 3D view are distorted and not readable.

[image]

### Expected results

Fonts rendered properly :)

### Project file & Debug log uploads

[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/12037319/BambuStudio.zip)

(Project file does not matter, just load any .stl file)

### Checklist of files to include

- [X] Log file
- [X] Project file

mzanetti, 2023-07-21:
Just found out that this only happens when running it on the integrated Intel graphics. Running it on the discrete nvidia GPU, all texts are rendered properly.

luciotorelli, 2025-06-29:
Same issue on Bazzite 41 (AMD Framework 13), still open and I am unable to run on discrete GPU :(