bambulab/BambuStudio #2041: Some geometry corrupting projects in 1.7.0 beta

RyanEwen, 2023-07-13:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Create a new project.
2. Add the attached STEP file model.
3. Save the project.
4. Close and reopen project.

I am not sure if this is limited to 1.7.0 beta or if it might also affect previous versions.

I have corrupted 2 projects and unsure how to recover them. Any tips would be appreciated.

### Actual results

[image]

### Expected results

Project open as normal.

### Project file & Debug log uploads

[Inner & Outer.zip](https://github.com/bambulab/BambuStudio/files/12042805/Inner.Outer.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DokuKaefer, 2023-07-13:
Problem is the **&** in the **_step_** file.
Remove it, open it, Save the project and it works.

RyanEwen, 2023-07-13:
Even if it the step file is the root cause, I think the slicer should still be able to reopen the project again, especially since it was able to create the project and print it and save it without issues or errors. Why does it treat it differently only after saving it?

DokuKaefer, 2023-07-13:
Good question, that must answer us a programmer.

bambu123, 2023-07-14:
Maybe @DokuKaefer is right. But this shoud be fixed in the 1.7 version. We'll check it again.

RyanEwen, 2023-07-14:
> Problem is the **&** in the **_step_** file. Workaround: Remove it, open it, Save the project and it works.

I can't seem to find the `&` in the `.step` file when opening it with VS Code. Can you show me where it is?

DokuKaefer, 2023-07-14:
> > Problem is the **&** in the **_step_** file. Workaround: Remove it, open it, Save the project and it works.
> 
> I can't seem to find the `&` in the `.step` file when opening it with VS Code. Can you show me where it is?

the & in the filename of the step file is the problem. 
Rename it like inner_outer.step

RyanEwen, 2023-07-14:
Ah I see. Looks like if I just rename it within the slicer by hitting F2 on it, that works as well
[image]

zhimin-zeng-bambulab, 2023-08-18:
It has been fixed in the latest version.
If you use the new version of the software to save the model file with & in the name, it can be opened normally with the new version.

closed 2023-08-18 by zhimin-zeng-bambulab