bambulab/BambuStudio #2044: Flatpak support

cortopy, 2023-07-13:
**Is your feature request related to a problem? Please describe.**
Linux versions are released for Ubuntu and Fedora but there are many more distributions like Manjaro

**Describe the solution you'd like**
Flatpak is a popular format to package apps for all linux distros

**Describe alternatives you've considered**
Install from Arch aur repository but it's broken. I can see some GitHub issues filed about this

**Additional context**
Supporting flatpak would simplify releasesas only one Linux would have to be supported. And would probably stop new GitHub issues about installing on several Linux distros

Notarin, 2023-07-14:
I've found that pretty much any distro im on, one of the two provided packages will work for me, for example, I use the fedora package on my arch install (btw)
But I do support a flatpak release, it would assist distros and stores that relyon the flatpak ecosystem

cortopy, 2023-07-14:
I opened this issue because I tried the Ubuntu one and didn't work. It would start, but then hang indefinitely. Fedora seems to work better, but still not sure. Thanks for the suggestion!

I opened this issue because having two flavours of AppImage suggests to me that neither one of them is truly multi  arch. I'm not a fan of flatpaks, but I've never had an issue with them. They run smoothly and can upgrade easily. 

My suggestion is also for the maintainers of Bambu Studio, as I'm guessing the burden on them could be reduced with having just one flatpak for all Linux distros. If I search for [issues related to Linux](https://github.com/bambulab/BambuStudio/issues?q=is%3Aissue+is%3Aopen+linux) there are currently 24 open!!

WayneManion, 2023-07-14:
AppImage is supposed to be distro-agnostic. Why are there two different AppImages for Fedora and Ubuntu?

craigcabrey, 2023-07-16:
Definitely interested in a Flatpak as well.

@WayneManion AppImage being distro agnostic is simply not true. We depend on the packagers of the AppImage to get it right. On my Silverblue installations, I still have launch a container and install a bunch of packages to make Bambu Studio work.  Lame.

WayneManion, 2023-07-16:
AppImage architecture allows for executable files to run on pretty much any Linux distro. The BambuStudio ones just aren't implemented very well.

craigcabrey, 2023-07-17:
Right, that's what I said by relying on the developers to implement it
correctly. Better to just use Flatpak where the forcing function is much
stronger.

MilesRdz, 2023-08-17:
Just wanted to chime in my support for this feature.
I'm having a hard time compiling bambustudio for Arch (segfault) and a flatpak would go a long way of supporting more Linux distros.

crtvrmn, 2023-08-31:
+1
for a random noob (like myself), pruca slicer is way easier to install

candrews, 2023-10-18:
PrusaSlicer has a flatpak: https://flathub.org/apps/com.prusa3d.PrusaSlicer

Since Bambu Studio is a fork of PrusaSlicer, I suspect a lot of the flatpak work can be reused.

watzon, 2023-10-29:
The appimage required some work to get running on Silverblue as well. Had to layer mesa-libOSMesa to get it to launch. A flatpak would be extremely nice.

delock, 2023-10-30:
+1.  Running Archlinux but AUR installation is broken.  Flatpak is really neat.

nokiomanz, 2023-12-16:
+1. Running NixOS on my side and a Flatpak would be amazing.

hadess, 2023-12-17:
I'm working on this, ETA is, hopefully, in time for the new year. I'm currently at more than 100 patches, 2 week-ends and loads of evenings to get it building.

hadess, 2023-12-29:
I've pushed my absolute mess of a Flatpak branch here: https://github.com/hadess/flathub/tree/wip/hadess/bambustudio-from-source
I'll need to bring in a lot of changes to the metadata and launcher that I made in another branch when I tried to base it off the AppImage.

And my build patches for BambuStudio are here for review: https://github.com/bambulab/BambuStudio/pull/3139

There's still issues with the default config not working to download the network plugin, and the video preview not displaying anything, at least. I'll post a PR for review on the Flathub repo as soon as those are fixed and the app is usable for me.

If you want to support me for the evenings and week-ends I've spent on it so far, and you can afford it, I have a [3D printing-related Amazon wishlist](https://www.amazon.fr/hz/wishlist/ls/3A4WBDMGAOEDS?type=wishlist&filter=unpurchased&sort=price-asc&viewType=list).

hadess, 2023-12-30:
> I'll post a PR for review on the Flathub repo as soon as those are fixed and the app is usable for me.

https://github.com/flathub/flathub/pull/4822 is available for testing. Please don't report bugs that are listed in the TODO list.

Notarin, 2023-12-30:
@hadess Thank you for your work on this

hadess, 2024-01-11:
https://flathub.org/apps/com.bambulab.BambuStudio is available.

@lanewei120 if you or anyone else at BambuLab wants to help maintain the application, please file a new issue at https://github.com/flathub/flathub/issues and cc me. This issue can be closed.

For the others, the wishlist mentioned above is still available :)

lanewei120, 2024-01-15:
> https://flathub.org/apps/com.bambulab.BambuStudio is available.
>
> @lanewei120 if you or anyone else at BambuLab wants to help maintain the application, please file a new issue at [...]

@hadess

thanks very much for doing this~

closed 2024-01-15 by lanewei120

blathers123, 2024-02-14:
@hadess Thanks a bunch for getting the flatpak to work. Can I buy you a beer? Your donate page goes to amazon france, which is giving me issues (based in the US)