bambulab/BambuStudio #2045: [bug] Tree supports do not work with support enforcers

Antyos, 2023-07-14:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel 11th gen i9-11900H
Memory: 64 GB
GPU: Nvidia GeForce RTX 3050 Ti

### Printer

Bambu Lab P1P

### How to reproduce

1. Load an STL file with an overhang
2. Add a support enforcer cube
3. Enable supports and set the type to `tree(auto)` or `tree(manual)`

[image]

### Actual results

The tree supports seem to create an enclosed shell around the part wherever the support enforcer is. The entire support enforcer mesh seems to be filled with support interface layers, while everything below it is regular rectilinear support. 

#### Tree support
[image]

Using normal support works as intended. So does using tree support with paint-on supports.

#### Normal support
[image]

### Expected results

Tree supports should only be generated under the parts of the model intersecting with the support enforcer, like with normal supports.

### Project file & Debug log uploads

[Overhang.zip](https://github.com/bambulab/BambuStudio/files/12045640/Overhang.zip)
[debug_Thu_Jul_13_20_45_54_21548.log](https://github.com/bambulab/BambuStudio/files/12045645/debug_Thu_Jul_13_20_45_54_21548.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-07-20:
Thanks for reporting the bug. I'll fix it ASAP.

tangketan, 2023-08-10:
Fix included in Studio V1.7.