bambulab/BambuStudio #2046: Layer Time Control and Visualization Improvements

cosmikwolf, 2023-07-14:
**Is your feature request related to a problem? Please describe.**
When I am doing a very detailed print, the time between layers affect the contraction of filament after deposition. I set the minimum time, but it is hard sometimes to see which layer exactly since it is just a gradient from pink to green.  I am trying to get perfect bridges with no support, and layer timing is very important for small prints. 

**Describe the solution you'd like**
Feature 1: It would be great if I was able to click the layer time colors, to highlight the layers which have that amount of time.
Additionally, it would be great to have a button to click to see which layers will actually be slower than indicated due to the min time specified in the filament cooling section. Currently, that setting does not change the layer times color scheme view

Feature 2: a button to set consistent timing from layer to layer, so that the time between the start from layer to layer is exactly the same every time. Like, each layer takes 17 seconds, or something like that, with the minimum being the longest running layer. A checkbox to make it either make all speeds slower, or to simply wait after the short run layers would be advantageous. 

Feature 3: A delay setting that could introduce an arbitrary delay from the finish of one layer, to the start of the next layer. This would be an easy fix to curled overhangs from improper cooling. 

**Describe alternatives you've considered**
None that I can think of.

**Additional context**
This screenshot shows what I am looking at. I have very small imperfections due to the difference in cooling time between layers, and would like to be able to isolate layers in order to better tune the timing of the different speeds. 

[image]

MadMax389, 2023-07-14:
You can see the actual layer time of each layer from the preview screen. Move the slider at the bottom to see the time. It certainly doesn't do all you are asking for but at least you can check individual layers.
[image]

cosmikwolf, 2023-07-14:
oh interesting! That certainly helps.  I think my suggestions would make it a bit more obvious, but it is good to know that I can get access to that info at a glance.

odysoftware, 2024-03-19:
But as yo can see from this screenshot and I verified it also in the current version, the layer time does not show the correct color in the legend. Just look at the above screenshot, the layer time from the legend would be around 1-3s but it is 24,5 so the color should be yellow to green...

Would be great if this could get fixed?

lanewei120, 2024-03-20:
@odysoftware could you share a 3mf to help us to debug

odysoftware, 2024-03-20:
> @odysoftware could you share a 3mf to help us to debug

Since this is not directly related to this open issue, I have created a bug report here: https://github.com/bambulab/BambuStudio/issues/3741 <-- there you can find a 3mf file and also see the problem.

Thanks a lot,
Oliver