bambulab/BambuStudio #2051: Studio 1.7.0.65 Crashing Randomly

MadMax389, 2023-07-14:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

10th Gen Core i5
32GB
RTX3060

### Printer

X1C and P1P

### How to reproduce

I have not been able to reproduce this, but it has happened five or six times since installing v 1.7.65. I have both X1C (2 AMS) and a P1P (1 AMS). I use a single instance of Studio to manage both printers, so I am always switching the device screen between printers. The crashes always seem to happen right when I select the other printer from the device screen. (I'm in the X1C beta program and am running FW 1.05.06.19 on the X1C)

### Actual results

For troubleshooting, I opened a new instance of Studio and switched printers 20 times, but did not crash.

### Expected results

No crashing. I did not experience this in 1.6

### Project file & Debug log uploads

Windows event log and BS logs for the most recent crash

[BS 1.7.0.65 Windows Application Error.zip](https://github.com/bambulab/BambuStudio/files/12050293/BS.1.7.0.65.Windows.Application.Error.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MadMax389, 2023-07-14:
It just did it again 2 minutes ago when switching from P1P to X1c on device screen. Updated bambu log files attached.
[Log.zip](https://github.com/bambulab/BambuStudio/files/12050638/Log.zip)

bambu123, 2023-07-17:
Please send me these files from your system.
- C:\Windows\System32\jscript.dll
- C:\Windows\System32\kernel32.dll
- C:\Windows\System32\wmp.dll
- C:\Windows\System32\ntdll.dll
- C:\Windows\System32\KERNELBASE.dll
- C:\Windows\System32\user32.dll

MadMax389, 2023-07-17:
[DLL.zip](https://github.com/bambulab/BambuStudio/files/12068763/DLL.zip)
As requested..

bambu123, 2023-07-17:
> [DLL.zip](https://github.com/bambulab/BambuStudio/files/12068763/DLL.zip) As requested..

Thank you. But the DLLs are not all consistent with the crash dump. Are you copy DLLs from the computer where the crash occurs?

MadMax389, 2023-07-17:
> > [DLL.zip](https://github.com/bambulab/BambuStudio/files/12068763/DLL.zip) As requested..
> 
> Thank you. But the DLLs are not all consistent with the crash dump. Are you copy DLLs from the computer where the crash occurs?

Yes, same system. I copied them out of the C:\Windows\System32 folder
[image]

MadMax389, 2023-07-17:
Last Windows updates were 5 days ago

[image]

MadMax389, 2023-07-17:
It just did it again. I was on the device screen with the P1P selected. It seemed to have lost the colors/materials assignments since all AMS slots had question marks. (I didn't get a screen shot). I selected the X1C and it crashed. Log attached.
[crash_Mon_Jul_17_10_44_51_0.log](https://github.com/bambulab/BambuStudio/files/12070688/crash_Mon_Jul_17_10_44_51_0.log)

bambu123, 2023-07-18:
> It just did it again. I was on the device screen with the P1P selected. It seemed to have lost the colors/materials assignments since all AMS slots had question marks. (I didn't get a screen shot). [...]

Please send all logs include the DMP files.

bambu123, 2023-07-18:
@MadMax389 
It is relate to live view. 
Start/stop live view have problems in your computer.
Is it the same with the 1.6 version.

MadMax389, 2023-07-18:
All log files
[log.zip](https://github.com/bambulab/BambuStudio/files/12074984/log.zip)
I have noticed at times that live view will freeze or not display at all when switching printers. I don't recall having the issue with 1.6, but I've only had the P1P for about four weeks, so I would only have been using Studio for 1 printer before then.

bambu123, 2023-08-21:
@MadMax389 Have you test the lastest 1.7.4 version? Is the problem still exists?

MadMax389, 2023-08-21:
@bambu123 If I don't enable the camera live feed, I don't seem to have the problem. I've been running with my camera off on both printers for a few weeks. I'll enable the camera and stress test over the next few days.

bambu123, 2023-08-21:
> @bambu123 If I don't enable the camera live feed, I don't seem to have the problem. I've been running with my camera off on both printers for a few weeks. I'll enable the camera and stress test over the next few days.

Okay, thank you.

MadMax389, 2023-09-02:
I have been testing this last week using one instance of Studio for both the P1P and the X1C and switching between printers. So far I have not had the program suddenly crash to desktop as it did before. However, sometimes the camera on the P1P will freeze showing the image from the last print. If I switch to the X1C and then back to the P1P, it seems to "unfreeze" the camera feed, and then acts normally. No crashes, however.