bambulab/BambuStudio #2052: Cannot Edit K Factor nor use Flow Dynamics Calibration

DamoclesSword, 2023-07-14:
### Bambu Studio Version

1.7.065

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go to Device Page
2. Select an AMS (I have two)
3. Click on a filament to edit 
4. Drop down menu for PA Profile does nothing and Factor K is greyed out. 
[image]

### Actual results

Drop down menu for PA Profile does nothing and Factor K is greyed out. Given that Dynamic Flow Calibration is currently locked behind a closed beta, I have zero means to calibrate my filament and will have to resort back to Orca. 

### Expected results

I am assuming PA Profile should be editable and that K Factor can be typed in. Right now, there is no way to calibrate or save a K value. 

I am in the middle of a large 100 order production run. All of my prints have holes from the issue. When updating to the beta 1.7, no where did it state that certain features are locked behind a firmware update that is not accessible. 

Bambu either needs to open the Flow Dynamics to the current release firmware, or release another beta version with the calibration options enabled 

### Project file & Debug log uploads

[K Factor Issue.zip](https://github.com/bambulab/BambuStudio/files/12052927/K.Factor.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MadMax389, 2023-07-14:
Maybe locked because the printer is busy?

DamoclesSword, 2023-07-14:
> Reference

The printers were not printing when I tested it. I have tried reinstalling which also did not help.

MadMax389, 2023-07-14:
Yes, I see that. I may be wrong but I am thinking you have to run the Flow Dynamics Test first, then select the filament type that is generated from the test. You can't change the k factor from the device screen, but you can modify it on the manage result screen. It's not going to work unless you set the filament type in the AMS to whatever type was generated in the flow dynamics test.  That's a pain for me because I set all my AMS filaments to Generic PLA. But I guess it won't matter what type of PLA is set in the AMS since it will use whatever filament profile you set in the slicer.

[image]
[image]

lanewei120, 2023-07-17:
@DamoclesSword this values can be editable through the history 
[image]

from device page
we can only select the history values
[image]

if there is no history calibration result(just as your picture)
we will use the default value 
maybe we can display the default calibration value in this case

DamoclesSword, 2023-07-17:
@lanewei120 I am assuming you have access to that because you have the closed beta software for the X1 Carbon. It seems that for the P1 series, there is an option to edit K Factor but for the X1, there is no means. 

Since I cannot run a calibration due to this feature being locked to the beta 1.6 printer firmware, I have no means to edit the "manage results" as no results have been generated. 

[image]

Even on the Wiki (found here: https://wiki.bambulab.com/en/software/bambu-studio/extrusion-calibration?fbclid=IwAR08soWF4PLbRYyjT9AaahCIXWPGse0a8mMcYS3419GEQ1Gsbvf86QEfL7k ) 

It states clear instructions for Extrusion Calibration but those features seem to be nonexistent. Furthermore, in the Wiki section on Flow Calibration ( https://wiki.bambulab.com/en/software/bambu-studio/calibration_pa ) it does not state that this feature is not open to public. 

Needless to say, I am just trying to use Bambu Slicer but my prints have too many defects without a proper means to calibrate the filament.

lanewei120, 2023-07-18:
> @lanewei120 I am assuming you have access to that because you have the closed beta software for the X1 Carbon. It seems that for the P1 series, there is an option to edit K Factor but for the X1, [...]

oops, thanks for the detailed information
I downgraded my printer to the same version with yours, and will check how to solve this issue

BTW
are you using X1C or P1P, or both?

DamoclesSword, 2023-07-19:
@lanewei120 I appreciate you checking on this. Why is this feature so widely pushed on the Wiki and even the documentation for the Beta software if it is only available for a small select group of people with beta printer firmware? 

I also do not understand why a cheaper printer (P1 series), has an option for manual K value that the X1 series does not. I know the Lidar is meant to be used for the X1, but that only works with certain filaments and build plates. It also creates variabilities if mass manufacturing parts over time (like I do). 

If someone could give me access to the beta firmware, I think I would be a good test candidate as I am printing hundreds of parts to ship per month. Right now, without access to the K Value I have to use a different slicer as all my prints have gaps.

SaltWei, 2023-07-19:
> If someone could give me access to the beta firmware, I think I would be a good test candidate as I am printing hundreds of parts to ship per month

https://wenjuan.feishu.cn/m/cfm?t=stwCukXWvLNi-4y24

DamoclesSword, 2023-07-19:
> > If someone could give me access to the beta firmware, I think I would be a good test candidate as I am printing hundreds of parts to ship per month
> 
> https://wenjuan.feishu.cn/m/cfm?t=stwCukXWvLNi-4y24 We reopen the applied channel for you. You can send request for beta test now.

Submitted. I appreciate that.

SaltWei, 2023-07-19:
> Submitted. I appreciate that.

What's the first several letters of you email? we need to identify and match.

DamoclesSword, 2023-07-19:
cross

SaltWei, 2023-07-19:
done. You can try beta firmware now

zhimin-zeng-bambulab, 2023-07-20:
The old firmware version does not support modifying k, we will hide this item, and the new firmware version will be able to use this function normally.  Thanks！

zhimin-zeng-bambulab, 2023-08-04:
It can be verified on version 1.7.1, with firmware version 1.6.0.0

closed 2023-08-18 by zhimin-zeng-bambulab

keithel, 2024-07-12:
I suppose this may be a new issue, but it's very closely related.

I have a P1P. I have happily, for more than a year, been able to manually set the K factor of filaments right from within the AMS slot edit dialog.

The dialog shown would show:
* Filament (dropdown picking filaments - only the defaults though - no custom filament profiles you make, which is very irritating - separate issue though)
* Color
* Nozzle Temperature Max and Min (uneditable - presumably that would be editable in the filament profile)
* Factor K

Recently, I went to a friend's house and wanted to print on his printer - an X1 Carbon. I logged out of my account, and logged into his account, made a 3MF for his printer and initiated print.

Now, a week later, logged out of his account and back into mine with P1P printer profiles, and now when I go to the device tab, AMS slot dialog, the bottom half of the dialog now shows "Factors of Flow Dynamics Calibration" (I can't recall if that wording was there before or not), and has a choice for "PA Profile" with a dropdown, that I did not have before.  The dropdown was empty until I did some manual calibrations from the Calibration tab, after which it auto-populated the PA profile for the slot I did the calibration on...  Now, the problem I have is that the "Factor K" edit box is uneditable - whereas before it was editable.

I want to directly edit this value. I can see tying it to a PA profile, and if one of those is selected, make the K factor uneditable, but there should be a choice to not pick any PA profile, and allow you to manually enter it, like I previously was able to do.

Another thing is that, while I can create new PA Profiles, there is no "Manage Result" button in the calibration tab, so I cannot edit or delete the profiles created.

zhimin-zeng-bambulab, 2024-07-12:
@keithel  Could you create a new github issue, describe your phenomenon, and write down your software version number and firmware version number.

keithel, 2024-07-24:
Yes, will do. It still affects me.