bambulab/BambuStudio #2053: Unknown/unused settings are removed from profiles on save

CoryCharlton, 2023-07-14:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

X1C or P1P

### How to reproduce

1. Edit a filament profile in Orca Slicer and enable pressure advance and set a value
2. Close Orca Slicer to let the profiles sync to the cloud
3. Edit the same filament profile in Bambu Studio and adjust flow rate (or any other setting)

### Actual results

The "enable_pressure_advance" and "pressure_advance" settings have been removed from the profile.

### Expected results

All settings that were not specifically modified should have the same values they had prior to editing the profile in Bambu Studio.

### Project file & Debug log uploads

No files are relevant to this issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-07-17:
You can edit these parameters in BambuStudio directly.
You should select a thirdparty filament. Bambu filaments have not these parameters.

[image]

rfield19, 2023-07-20:
The pressure advance option you show above is not available in bambu studio 1.7 b2.  
This is with advanced, and developer mode switched on.  
Doesn't matter if it's a generic profile or not, the option simply does not appear.
[image]

CoryCharlton, 2023-08-08:
> You are in BBL P1P. Select a thirdparty printer preset will bring it to front.

Just to be clear. This issue is not about whether the setting appears in the UI or is used during printing.

The issue is that if you follow the steps to reproduce and then open the JSON file stored at `%APPDATA%\Roaming\BambuStudio\user\<USERID>\filament` the setting value has been removed from the profile entirely.

bambu123, 2023-08-09:
> > You are in BBL P1P. Select a thirdparty printer preset will bring it to front.
> 
> Just to be clear. This issue is not about whether the setting appears in the UI or is used during printing.
> 
> The issue is that if you follow the steps to reproduce and then open the JSON file stored at `%APPDATA%\Roaming\BambuStudio\user\<USERID>\filament` the setting values have been removed from the profile entirely.

I see. Could you please send me the preset file before you edit in BambuStudio.  That is `%APPDATA%\Roaming\BambuStudio\user\<USERID>\filament\xxx.json`. 
One more thing, If the config value is the same as the value in the inherits preset, it is removed from the saved json.

CoryCharlton, 2023-08-09:
Well I am no longer able to reproduce the issue so I'm going to close this for now.

closed 2023-08-09 by CoryCharlton

bambu123, 2023-08-09:
> Well I am no longer able to reproduce the issue so I'm going to close this for now.

OK. If you have any questions, please continue to contact us.