bambulab/BambuStudio #2054: Calibration tab - forgetting selected printer after fresh start

serkanaksu, 2023-07-14:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C with AMS

### How to reproduce

1. Go to Calibration
2. Check on left side, it says "No printer"
3. Click on "No printer"
4. Select your printer below

### Actual results

No automatic printer selection like in Device tab.
Printer needs to be selected after fresh start of Bambu Studio

### Expected results

Last printer should be selected automatically like in Device tab

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12053270/log.zip)
[Untitled.zip](https://github.com/bambulab/BambuStudio/files/12053273/Untitled.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-07-18:
Hi. @serkanaksu we are doing this

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambus Studio

closed 2023-10-13 by walterwongbbl