bambulab/BambuStudio #2056: Copy slicer settings from one part/object to another

RyanEwen, 2023-07-15:
**Is your feature request related to a problem? Please describe.**
Sometimes I am unable to replace a model in a project with an updated version of the model, so I have to add the new model and then manually transfer the slicer settings from the old model to the new model. This is very cumbersome when the model contains many (7) parts each requiring different settings, and then the bed is filled with many instances of these models. 

**Describe the solution you'd like**
I would like the ability to copy the custom settings from one part/object onto another part/object. I think it could be an option in the context menu.

**Describe alternatives you've considered**
If I could replace a model containing multiple parts with a new version of that model, I wouldn't need this feature, but I still think it's a useful feature for other use cases as well.

phntsm, 2024-05-29:
Coming from trying to find out if this was even possible for about half an hour and finding this, I whole heartedly support this feature request.

kengregson, 2024-07-13:
+1  Would save time to copy/apply custom part specific settings when I update and import new stl for one part on a plate

RyanEwen, 2024-10-15:
> Would save time to copy/apply custom part specific settings when I update and import new stl for one part on a plate, Actually,....for anyone landing here, right-click on selected object->Replace with STL does exactly that.

This only works for simple objects and not multi-part objects.