bambulab/BambuStudio #2057: Currently, there is no option to adjust the first layer flow.

wonsuhna1, 2023-07-15:
There is no option to adjust the first layer flow.
OrcaSlicer is a "Bottom surface flow ratio" option, but this is an adjustment option for the first layer "solid infill"
I don't see a first layer wall flow ratio adjustment, can you add one?

This is needed to in "Non bambu printers", adjust the z-axis size of the print. (z axis printing size fine tune.)
When I set the first layer height to 0.2, I really want the first layer to be 0.2mm high.
To do this, I need to be able to adjust the flow ratio of the first layer.

maybe, some people needed in bambu printers.

SaltWei, 2023-07-17:
@wonsuhna1 Thanks!

@QingZhangBambu this should be wrong code in gcode.cpp. This setting should be applied to initial layer which is contacted with the hot bed, rather than erBottomSurface. erBottomSurface may not be contacted with hot bed.

[image]

SaltWei, 2023-07-17:
@BambooBugFixer Please fix this issue.
You can ask @QingZhangBambu for more detail about this ticket after you understand user requirement yourself.

BambooBugFixer, 2023-10-23:
This feature has been available in the newest Studio version.

GuzziRaz, 2023-11-05:
[image]

@DriftMonkey100 you need to enable Developer Mode in settings.

SaltWei, 2024-01-18:
Already included in latest version. Thanks for reporting issue to us！

closed 2024-01-18 by SaltWei