bambulab/BambuStudio #2058: Support Interface slicing bug:   Extra interface placed improperly in middle of support structure, causing print failures

psiberfunk, 2023-07-15:
### Bambu Studio Version

1.7.0.65

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

MacOS

### Printer

X1C 

### How to reproduce

Open Attached 3MF project file.  Slice.  

Notes:

- Manual painted supports are being used to isolate this bug, and I have ensured the only part selected for support is the overhang.  Nothing in the area of the bug is supposed to be supported.
- 

Screenshots

### Actual results

Screenshots here show what happens with the slice, and that the actual print also does this, improperly.  Because this interface layer is intentionally weak it causes the whole support to break and the print to fail.

[image]

[image]

[image]

### Expected results

The slicer should not be generating an interface / filament switch at this point in the support.  This is clearly a slicer bug.

### Project file & Debug log uploads

Project file attached.

Log file N/A because this is a slicer logic bug.
[SupportInterfaceBug.3mf.zip](https://github.com/bambulab/BambuStudio/files/12061473/SupportInterfaceBug.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-07-20:
@psiberfunk This is actually not a bug. We have a rule that support material can't be in the support base. Any other material is allowed. This is to ensure the color changing is minimized. 
Set the interface filament to any _support_ material and you'll see everything is fine.
[image]

tangketan, 2023-07-20:
If you want to use a non-support material as interface, there are two ways:
1. enter develop mode, set Support material in the filament setting
[image]

2. specifically set the filament of support base
[image]

psiberfunk, 2023-07-20:
> @psiberfunk This is actually not a bug. We have a rule that support material can't be in the support base. Any other material is allowed. This is to ensure the color changing is minimized. Set the interface filament to any _support_ material and you'll see everything is fine. [image]

How is this not a bug though?  it's randomly inserting something other than the default material.  The need to set it to a "support" material that's defined explicitly as a support is kind of silly and totally not clear to the user at all.  It clearly forces the user to think way too hard and clone a material profile.. or set it explicitly.

It's not at all clear that you should have to set the base/raft material explicitly here.  I would suggest that the "default" needs to be changed to "Any filament allowed" to be clear to the user what's going on.  As it is, it's not at all clear.

psiberfunk, 2023-07-20:
Furthermore, while "any" is reasonable when the materials are the same TYPE of filament.. when they're different filaments like PETG vs PLA, it's clearly always a bad idea and should be prevented.  I still maintain the behavior as it is is a bug that will lead to print failures.

psiberfunk, 2023-07-20:
> If you want to use a non-support material as interface, there are two ways:
> 
>     1. enter develop mode, set Support material in the filament setting
>        [image]
> 
>     2. specifically set the filament of support base
>        [image]

Having to use develop mode isn't a valid fix.  Fixes should not be buried behind developer options, from a UX perspective.  The other fix is valid but completely obtuse and not at all easy to discover.

psiberfunk, 2023-07-20:
P.S.  If you've decided, despite my logic above, not to fix this, you should close with "won't fix".   My recommended solution is to:

1. Relabel "default" to "Any color PLA/PETG/etc"
2. Change the code to force the support filament (within the main support body) to stay the same TYPE (PLA, PETG, etc).
3. Do the same as 1. & 2., but for also for the support interface, as well.

tangketan, 2023-08-02:
By "not a bug" I mean it's totally following the design purpose, not saying it's optimal.
And I do agree with some of your suggestions, which would certainly make things clearer. I'll discuss it with our team.

jennyy867, 2023-08-23:
I'm experiencing the same issue, and ran through probably two rolls of filament before realizing it was the slicer update that changed. I have since had to roll back to an older version to keep printing for my business. I'm seeing the extra support interface not just on trees as the image above, but a whole interface layer in PLA on my model before the true support interface starts with Support for PLA. These filament lines cannot be removed making my models unsellable. 

Is there a reason this cannot be changed back to the functionality before the update? Or a timeline for a new release? I'm not able to use the new calibration features, which is a bummer, until this is resolved in a new release. This works properly in Bambu_Studio_mac-v01.06.02.04. and I confirmed today it is still broken in Bambu_Studio_mac-v01.07.04.52.

[image]
[image]

jennyy867, 2023-08-23:
And I changed it to white (model is usually black) so you could see, but it may be confusing since the support color is white. Here's the model sliced in blue with white support interface as well so you can rule out it working but looking the same color in the slicer :)

[image]

Moeser, 2024-02-08:
This looks like an actual bug.  Is @tangketan understanding the issue correctly?

> This is actually not a bug. We have a rule that support material can't be in the support base. Any other material is allowed. This is to ensure the color changing is minimized.

So are you saying the rule is causing this behavior, or that the rule is a way to fix this behavior?  **The support interface material is being used to print things other than the support interface**.  Why is that not a bug?  To be clear: When I'm printing with PLA, and I **select PETG only for the support interface**, it starts using PETG for areas that **are not part of the support interface**.  

Why is Bambu Studio using the support interface material for things other than the support interface?  Do you really expect this to be the correct non-buggy behavior?  This is a bug, and it's ruining prints.  Please don't blame your users for this problem.

[image]
[image]