bambulab/BambuStudio #2059: Question/potential support - Does this printer have power loss recovery?

Majestic7979, 2023-07-15:
### Bambu Studio Version

1.7 beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Not relevant

### Printer

X1 Carbon Combo

### How to reproduce

This is more of a question, and depending on the answer **FROM BAMBU LAB** I will see if I need support.

Some printers have "power loss recovery" - Let's take a benchy as an example... you send the benchy for print, let's say it has 600 layers... the printer is doing the print fine, it's currently at layer 486, so there are still some 114 layers to finish, when all of a sudden the power is cut. The dog pulled the lead from the socket. Or maybe there was a lightning and the transformer blew up on the electric pole on the street. Or maybe there was a blackout in the area... Anyway, now the printer has stopped, and your benchy was almost completed.

So the power comes back 5 seconds later, and...... nothing. You've got a ruined benchy. The X1C does not resume the print from the last point when the power was cut.

Let's take the printer that I absolutely dislike because it's old and ridiculously overpriced (this is my opinion I'm not asking anyone to agree or fight about it) - Prusa Mk3S+/Prusa Mk4. This printer will resume the benchy after the power cut. It will continue to print the benchy as if nothing happened... Apparently it uses whatever is left inside the capacitors on the mainboard as a short lasting "battery" and it can save the current gcode in memory, so that when the power is back up it knows the last state and can continue from there. It can sense when the main power from the AC plug is cut, and use that last bit of power on the mainboard capacitor to save state before it shuts down a second later. This gives the Prusa just enough time to "remember" what it was doing, so it can continue once the power is back on. It's rather ingenious.

It does not justify to me the high price tag for their inferior, slow and outdated cartesian printer, but it's a good feature that could save some prints. It would not work for anything that requires a hot bed (PEI) as once the PEI plate cools down the print will detach... but if the power loss is momentary, let's say just a minute maybe it could be enough to save the print! So now the benchy can continue from layer 486 and be finished instead of an unfinished benchy causing waste and going in the garbage. I presume the X1C has capacitors in the board? Could this be programmed with firmware for the X1C?

When I bought the X1 Carbon Combo I thought such an amazing, advanced printer would have this issue. But apparently, it does not!?!?!?

It has something called "recover print", which IS NOT the same thing as I've come to find out after testing. This feature is for when the nozzle/print head loses track of X/Y, so it will rehome itself to reorient the coordinate and continue printing. This is to avoid layer shifts because of print collisions.

So I need to know from Bambu Lab DIRECTLY - not users reading this because they don't know the code running in the printer. Bambu Lab this question is for your engineers: DOES THE X1 CARBON HAVE "POWER LOSS RECOVERY"? Because if I try to print a benchy right now, and then in the middle of the print I pull the plug to simulate power loss, when I turn the printer back on there is no option to resume print, I can replicate this over and over. Printer is on latest firmware I think it's 1.05.02.

### Actual results

See above

### Expected results

I want the printer to resume from the same point it was when the power was lost.

### Project file & Debug log uploads

Not relevant as yet - ask if you need it.

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-07-17:
This is github for BambuStudio not for printer issue. We can transfer the issue to printer team internally.

For more rapid support for printer, you can raise a tickets by app and upload the printer log.
please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

Majestic7979, 2023-07-17:
> This is github for BambuStudio not for printer issue. We can transfer the issue to printer team internally.
>
> For more rapid support for printer, you can raise a tickets by app and upload the printer [...]

I understand it's for B-Studio but it's the easiest place to reach the development team and I know you can also accept Printer issues hence the yellow Printer label. And because your source code is closed I don't know if power loss recovery is even included and if it depends on the slicer/Bambu Studio or not so I am raising the question here.

I need an answer, let me know.

samcooler, 2023-07-28:
It does have power loss recovery. I just used it today. When the power comes back, hit Play.

Majestic7979, 2023-08-03:
> The P1P does have power loss recovery. I just used it today. When the power comes back, hit Play.

That's cool but my question is specific to X1 Carbon Combo