bambulab/BambuStudio #2061: Plate names are lost when using Undo

RyanEwen, 2023-07-15:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Name a plate. 
2. Place a model on the plate.
3. CTRL-Z or click Undo.

Or with an existing project such as the attached project:
1. Move the cube.
2. CTRL-Z or click Undo.

### Actual results

Plate names are erased.

### Expected results

Plate names remain.

### Project file & Debug log uploads

[Cube.zip](https://github.com/bambulab/BambuStudio/files/12061802/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2023-07-18:
Thank you for your discovery. It has been fixed. Stay tuned

SaltWei, 2024-01-16:
Please reopen if still has issue

closed 2024-01-16 by SaltWei