bambulab/BambuStudio #2064: X1C - Ability to change printer image on home screen

projectcobalt, 2023-07-16:
**Is your feature request related to a problem? Please describe.**
i love looking at the printer and its specs, but once you memorize them, the image becomes more of an advertisement and annoying. personalization is encouraged so this would be a good place

**Describe the solution you'd like**
ability to change printer image on idle home screen and smaller image on home screen during printing, Bambu Lab X1 Series can remain as watermark

**Describe alternatives you've considered**
none available, locked

**Additional context**
Add any other context or screenshots about the feature request here.

Majestic7979, 2023-07-18:
Would be cool if they allowed to set a name for the printer. So people wouldn't have to put labels all over them. Maybe the name could substitute the image. But make it optional. I actually like the original graphics.

projectcobalt, 2024-01-25:
Was this added to the firmware feature roadmap?