bambulab/BambuStudio #2065: Watching camera causes a crash

redlinebikeparts, 2023-07-16:
Currently using the latest version of the Beta of Bambu Studio. 

When watching the camera too long it causes Bambu Studio to crash this is with both of my P1P's. 

If I am watching the camera for a period of time or if the print finishes while watching the camera Bambu Studio crashes

bambu123, 2023-07-19:
@redlinebikeparts Please send me the logs include the .dmp files in the log folder.

WayneManion, 2023-07-20:
I have never been able to get the camera stream to work in BambuStudio (or OrcaSlicer). I have a P1P and I run Arch Linux. I have tried the Ubuntu AppImage, the Fedora AppImage, and the Arch User Repo (AUR) version of BambuStudio (built from source). I have also tried the different OrceSlicer AppImages and the build-from-source and precompiled binary versions available in AUR. 

I am currently running the Fedora AppImage 1.7.0 Beta 2. When I try to view the camera feed the app crashes immediately. In past versions the stream has always failed, but the window has typically stayed open. 

I didn't see any .dmp files in ~/.config/BambuStudio/log. The log files from today are here:

https://pastebin.com/3ACL8qjE
https://pastebin.com/RsEWgYCv
https://pastebin.com/mPNMcyeF
https://pastebin.com/wJUJG3mw

redlinebikeparts, 2023-07-23:
> @redlinebikeparts Please send me the logs include the .dmp files in the log folder.

how do i do that?

bambu123, 2023-07-23:
You can arrive at the log folder from the menu "Help > Open configuration folder".

bambu123, 2023-07-24:
@redlinebikeparts
I think you are in windows system.
For linux, please help debug with gdb, send back the crash stack.

To do:

$ gdb Bambu_Studio_linux_ubuntu_vXXX.AppImage
run
bt

please input the 'run' command in gdb. then when it stops on a crash, input the 'bt' command.
then send the console output back here.

WayneManion, 2023-07-29:
I tried my best to follow the instructions. 

Captured console log output: https://pastebin.com/ck42fd5G

This came from opening the app, switching to the Device tab, then clicking the Play button for the camera.

bambu123, 2023-07-31:
It seems to be an issue with the jpeg library which is not compatible.

Thread 199 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffdcedc76c0 (LWP 41841)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:168
168		jmpq	*%rdx
[?2004h[?2004l
[?2004h(gdb) bt
[?2004l
#0  ____longjmp_chk ()
    at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:168
#1  0xb3179467f25fcdd8 in  ()

lanewei120, 2023-07-31:
the same as https://github.com/bambulab/BambuStudio/issues/1814
and https://github.com/bambulab/BambuStudio/issues/1665#issuecomment-1550380137

@WayneManion @redlinebikeparts 

could you have a check in the above issues?

redlinebikeparts, 2023-07-31:
[image]

Which file to put here?

WayneManion, 2023-08-01:
2I am running Arch. I am running the AUR package (built from source) from here: https://aur.archlinux.org/packages/bambustudio-git

The application doesn't crash when I click the play button to view the camera stream, but I can't see any video either. It just says:

Loading Failed [2]!

The gdb logs from running the built-from-source version of the app is here: https://pastebin.com/bhUK8JAh

Polsaker, 2023-08-30:
I'm seeing the same error in Arch Linux with the latest fedora appimage. Here's what i managed to capture with gdb:

```
Thread 278 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffda0bff6c0 (LWP 1584553)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:168
Downloading source file /usr/src/debug/glibc/glibc/debug/../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S
168             jmpq    *%rdx                                                                             
(gdb) bt
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:168
#1  0x53e4ef590606223a in  ()
```

It happens as soon as I try to start watching the camera feed.

danexpo, 2024-10-22:
Same issue under Fedora 40 KDE for me.

IetIesAai, 2024-10-28:
same with fedora 40 gnome

aelsayd, 2024-10-29:
Same issue on Manjaro i3wm

bambu123, 2024-10-30:
Please use the new 1.10 beta version.