bambulab/BambuStudio #2066: Print Farm Capabilities

redlinebikeparts, 2023-07-17:
It would be amazing to have a print farm type capability and overview of printers in the device menu. 

When managing multiple printers, it's cumbersome to switch between them to find time remaining on prints, etc..

cbenhagen, 2024-07-12:
@itsonlymedan I guess this issue can be closed now that this is implemented. For potential improvements on the existing feature, please create new tickets.

itsonlymedan, 2024-07-26:
> @itsonlymedan I guess this issue can be closed now that this is implemented. For potential improvements on the existing feature, please create new tickets.

No im talking about a summary page. Easy to view what printers are running and what they are up to