bambulab/BambuStudio #2067: Dropdown menus do not work at all

jhogendorn, 2023-07-17:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.4.1 (22F82)

### Additional system information

Apple MBP M2 Max 32gb ram

### Printer

Bambu X1C

### How to reproduce

1. Install and open Bambu Studio
2. Attempt to use any dropdown menu, the item you select will be ignored and the value will not change

### Actual results

The setting is not changed

### Expected results

The setting should change

### Project file & Debug log uploads

[Body1.3mf.zip](https://github.com/bambulab/BambuStudio/files/12067399/Body1.3mf.zip)
[Archive.zip](https://github.com/bambulab/BambuStudio/files/12067401/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-07-18:
@jhogendorn Could you please test the lastest version?

jhogendorn, 2023-07-18:
its still an issue on the latest beta @bambu123

bambu123, 2023-07-18:
> its still an issue on the latest beta @bambu123

I'm not sure where is the dropdown menu.
Please provide a snapshot of the window.

jhogendorn, 2023-07-18:
literally any dropdown. selecting the printer, the bed type, any slicing settings that are a dropdown. any dropdown at all.

I can't use the printer because of this.

bambu123, 2023-07-18:
Ok. I see.
Can you use select preset using the keyboard Up/Down?

jhogendorn, 2023-07-18:
https://github.com/bambulab/BambuStudio/assets/253048/b1574de3-f116-4b24-a7d5-15297d918693

selecting a dropdown and using the arrow keys does work. this is a workaround for the moment thanks.

bambu123, 2023-07-18:
> bambustudio.mp4 
> selecting a dropdown and using the arrow keys does work. this is a workaround for the moment thanks.

I may relate to the macOS system, but I can't reproduce it in my macOS also with the version 13.4.1  (22F82).
Have you changed some settings in the System Settings of macOS?

jhogendorn, 2023-07-18:
no, this is a nearly brand new stock MacBook Pro m2 max. there's no odd or weird system customisations. I've tried running the app with rosetta and without and no difference.

jhogendorn, 2023-09-04:
Hi, is there any progress on this? I ask because it's a huge pain in the butt, and it actively prevents me from using the printer in a lot of ways. for example I can't change between presets if I have changed settings I want to keep because it reorders the list, and interrupts with a dialog, so hitting down means I can only toggle between two adjacent profiles.

bambu123, 2023-09-04:
Sorry, not yet. If there is any new news, we will let you know right away.

jhogendorn, 2023-09-04:
what can I do to expedite this or help debug?

jhogendorn, 2023-09-06:
I dug into this again. I disabled the better mouse software I run for external mice and drop-downs worked. I had done this prior as part of debugging so I'm not sure what the difference is. I dug through the bettermouse settings and it was specifically the 'click through' option enabled that caused dropdowns to fail. interestingly the built in macos clickthrough (using command while left clicking on an unfocused window) does not cause the same behaviour.

I'm happy for the bug report to be closed if you want, this feels like it might be a bug for bettermouse rather than bambu studio.

Apologies for the run about, I had tested this before lodging the issue so I'm not sure what the difference is. Perhaps just a matter of different time, a few reboots later, etc etc.

closed 2023-10-13 by walterwongbbl

rmustard, 2024-09-24:
I had this same issue.  And was trying to figure it out for weeks.  Same exact problem and solution, finally found that Better Mouse click through feature is the "problem".  Although I don't have issues with other drop downs in other software.

I tried to build BambuStudio to look into it, but it no longer builds on 14.6, and I don't have enough knowledge to help fix the dependencies causing the build break.

rickyzhangca, 2024-11-27:
exact same issue on v1.10.1.50, macos 15.1.1

imaggg, 2026-01-11:
@rmustard Thank you! Have same problem. Add exception

[image]