bambulab/BambuStudio #2069: Resize humidity sensor view for AMS in device tab

atl285, 2023-07-17:
**Is your feature request related to a problem? Please describe.**
In the current implementation the humidity view for the AMS is very small. It's hard to detect the current state.
[image]

**Describe the solution you'd like**
A bigger picture of the current state will be more usable and a big improvement.
[image]

**Describe alternatives you've considered**
As an alternative, you can show the current / real  state of the humidity sensor within the tool tip overlay.

**Additional context**
- none -

Waffull, 2023-07-21:
Please do this, and preferably change it to a % indicator...  Many of us are color blind and even if larger, can't tell what it's indicating.

A lot of useful material here: https://towardsdatascience.com/two-simple-steps-to-create-colorblind-friendly-data-visualizations-2ed781a167ec

nicoodeimos, 2023-07-29:
Same here, I'm colour-blind and struggle to understand the color system of the humidity icon.
Please add a text with % and "status" (poor, acceptable, ...) clearly indicated. thanks !

walterwongbbl, 2023-08-03:
@nicoodeimos @Waffull 
Sorry, we will adjust the humidity display. Use text or numbers to represent humidity values

timperlabs, 2024-05-25:
I noticed that after the recent update, the humidity icon is much better and is a number and large.   Can you provide a mapping between the 1-5 range and relative humidity values?