bambulab/BambuStudio #2071: New top and bottom surface patterns to create multicolor patterns with coextrusion filaments

VeilDango, 2023-07-17:
**Feature request intro**
Dual and tricolor filaments contain multiple colors on every strand, and a different color or blend of two colors will be visible on outer surfaces depending on the direction and angle of line printing. There are only a few top and bottom surface patterns in Bambu Studio that work well with dual and tricolor coextrusion filaments to make complex multicolor surface patterns, such as the Octagram Spiral pattern. With Octagram Spiral, some sections are actually either blends or alternating stripes of two colors, and different sections reflect light at different angles (especially if printed below 70mm/s), resulting in a unique effect by combining surface pattern and coextrusion filament.

If there were more patterns to choose from, it could be a very easy way for even novices to add comples patterns and colors to the top and bottom surfaces of their print. All they need to do is use a surface pattern with many angles and coextrusion filament. It is also possible to switch to a different filament after printing the first layer or two with coextrusion filament, so that a single 1kg spool could give hundreds of prints beautiful surface patterns.

**Solution**
There are other geometrical patterns that could greatly increase the possible top and bottom surface patterns and effects when used with coextrusion filaments. For star polygon patterns, there could be hexagram spiral, pentagram spiral, dodecagram spiral, etc. For tesselation patterns, there could be triangles, hexagons, checkerboard, 3D isometric cubes, etc.

There could even be complicated patterns like Celtics knots, floral laces, snow crystals, and cymatic patterns.

There could also be the option for people to import their own surface patterns into Bambu Studio, reducing or eliminating the need for Bambu Lab to create surface patterns.

**Alternative**
There could be a third-party plug-in tool that lets people add new surface patterns into Bambu Studio.

**Additional context**
Here are some examples of orange, blue, and green tricolor coextrusion filaments prints:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

QingZhangBambu, 2023-07-19:
yep, this feature already in our feature pool

QingZhangBambu, 2023-08-01:
#1463