bambulab/BambuStudio #2073: Option to lock custom settings in some menus - like height range modifier.

werewolfenn, 2023-07-18:
I like to run my first few layers nice and slow. At first I asked here for an option to choose a custom number of slower layers, (instead of just the first layer) and was pointed to Height Range Modifier.  This was great!   But it is a bit lacking.
WHen I choose this option, I can enter the speeds I wish to use for every option. That is great.   BUT, it won't stay set.   The next modifier I add, I have to do it once again.  Over and over.... it's annoying.   If I have a few dozen models on the bed, that aren't copies of others, I have to do this so many more times.  
Can something be done to lock in our preferences for areas like that?  Would certainly make things more comfortable.

lanewei120, 2023-07-25:
hello @werewolfenn

I can not understand your requirement well
after set the speeds in height range, these values should be kept

do you mean the values set in height range have been lost after add a modifier?

could you record a gif for your issue

werewolfenn, 2023-07-25:
> 

I will try to record something for you.

But yes,  When I make a modifier, and make the changes to the options (like speed) that I want, it will keep for that modifier only.  The next one I make is back to default settings.

werewolfenn, 2023-07-25:
[](url)

https://github.com/bambulab/BambuStudio/assets/120329025/1228beef-c227-4d4d-bb1a-812fa68e753c

lanewei120, 2023-07-26:
@werewolfenn
 hello, these three objects are seperate model objects
their settings are seperate, and their height range setting are also seperate

if you want to use one height range setting for all these objects
you need to merge them to one
then the height range setting will be shared
[image]
[image]

werewolfenn, 2023-07-26:
Yes, you are right.   BUT, what would be nice, is if the drop down menu would let me keep using the same variables, or settings from a previous use.  In filament settings, or profile settings, we can save any changes to the default settings, and they will remain each time we go to use them.  Can't this area get a way to save to make them personal?

lanewei120, 2023-07-26:
if we can copy the settings to others
it should be satisfy your requirement, right?

we are considering a similiar solution

werewolfenn, 2023-07-26:
Yes!   I don't mind doing a modifier for each use. I just don;t want to have to always need to change every number each time I use it.
Thank you