bambulab/BambuStudio #2074: Remote e-stop or reset urgently needed

cscashby, 2023-07-18:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Additional system information

n/a

### Printer

Bambu Lab X1C (and all)

### How to reproduce

1. Load AMS
2. AMS load fails, at 'loading filament'
3. Nozzle stays at 250degC with no way of resetting or emergency stopping, seemingly no timeout

### Actual results

Nozzle stays at temperature and no user way of overriding this

### Expected results

A timeout or failure of the AMS should result in printer being in a safe state

### Project file & Debug log uploads

[Studio logs.zip](https://github.com/bambulab/BambuStudio/files/12078430/Studio.logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

sodachen22, 2023-07-18:
Thanks for the feedback. It is a reason reasonable requirement especailly when the user is not near the printer. We will try to have a solution for this case.

Majestic7979, 2023-07-18:
Like Marlin's M112 command? Yeah, definitely.

cscashby, 2023-07-18:
Agreed, that's exactly what I'm thinking. All my printers (Klipper / Marlin / Octoprint) can do this as standard, and it's a bit of a shock this isn't built into Bambu.

Majestic7979, 2023-07-18:
> Agreed, that's exactly what I'm thinking. All my printers (Klipper / Marlin / Octoprint) can do this as standard, and it's a bit of a shock this isn't built into Bambu.

Oh you're in the UK too, hey hey!
Yep, pretty shocking I agree. If a bit part gets detached and can't fall through the gaps it could bend the hot-end badly. They need to implement this for sure. Even a few extra seconds could be the difference.