bambulab/BambuStudio #2075: Switch to next AMS spool when filament is empty

MrD-RC, 2023-07-18:
It would be a nice feature if you could switch to the next (or specified) slot, when the filament in the current spool runs out. This should be optional, and disabled by default.

For example, I'm printing a large project, which will use up the remaining filament on the spool in slot 1. It would be nice to be able to specify a slot to automatically use once the current spool empties. A simpler option would be to just use the next slot. But selecting a slot would be much more customisable. Maybe you could even present the option if Bambu Studio knows the print uses 700g of filament, for example. But the selected slot has less filament available on it.

Hologos, 2023-07-19:
This feature is already in there, it's called auto switch filament.

MadMax389, 2023-07-22:
> This feature is already in there, it's called auto switch filament.

Yes, but only if you remember to set up the AMS beforehand. It would be nice, as OP requested, to be prompted to switch to any slot with the same material when it runs out.

Hologos, 2023-07-22:
@MadMax389 Sorry, somehow missed that. +1 it would be cool

ChrisVonFrosta, 2024-06-19:
1 year later. Somthing happend? I still run in this error each role. But from remote i can't resume the print, even not with other color