bambulab/BambuStudio #2078: filament painting doesnt work properly

pietervisser05, 2023-07-18:
### Bambu Studio Version

1.7.0.53

### Where is the application from?

Bambu Lab github releases

### OS version

macos 13.4.1

### Printer

X1C

### How to reproduce

i paint my model in the prepare tab.
the whole body black, the letters "MW65" on top in yellow. after finishing painting it looks axactly how its supposed to print. however after slicing, the yellow colour disappears and the whole model prints in black

### Actual results

[image]
[image]

### Expected results

the letters should print in the colour i paint them

### Project file & Debug log uploads

[testprint.3mf.zip](https://github.com/bambulab/BambuStudio/files/12088671/testprint.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MadMax389, 2023-07-20:
You have several things going on here. 
1) Don't paint the letters, assign them colors instead.
[image]
2) The text shows up in this case
[image]
3) But then the text on the side walls doesn't print. They are too thin for the nozzle width at .2mm. You need to make it at least .4mm width, and then make sure they are in the proper order in the assembly. All the text objects must appear after the main object. In the pic below, I made the side wall text .4mm and arranged them to be last in the assembly list.
[image]

MadMax389, 2023-07-20:
To Bambu Lab: Regarding the order of the components in an assembly, it would be nice to be able to adjust the order by dragging the object in the list (like Prusa Slicer). In the pic below, I have two different assemblies with text flush to the top of the object. The only difference is the order in which they appear in the assembly.

[image]

Upon slicing, only the assembly where the text is last in the list slices correctly.

[image]

This is another result of my issue https://github.com/bambulab/BambuStudio/issues/1896

pietervisser05, 2023-07-21:
> You have several things going on here.
>
> 1. Don't paint the letters, assign them colors instead.
>    [image]
> 2. The text shows up in this case
>    [image]
> 3. But then the text on the side walls doesn't [...]

hello, thanks for helping me! regarding number 3, do you mean i should extrude the texts at least 0.4 mm from the object? or the letters itself should be 0.4 mm wide? i need the text to be flat, so extruding is not an option, i extruded the text now 3 mm into the body. i designed the model again in fusion 360, there in the assembly the solid body is first. however if i export it as step and load it into bambu studio the solid body appears last, so it doesnt slice correctly. do you know how i can change the order somehow?
greets pieter

MadMax389, 2023-07-21:
Yes, it's fairly simple. Select the text object in the object window and cntrl-c to copy and cntrl-v to paste. A copy will move to the bottom of the list. Delete the original. Repeat until the body is first in the list.

pietervisser05, 2023-07-22:
thanks a lot, that fixed it for me!

closed 2023-07-22 by pietervisser05