bambulab/BambuStudio #2079: Per Plate settings

moswald14, 2023-07-18:
**Is your feature request related to a problem? Please describe.**
Trying to keep Projects under one file but some models require spiral vase mode and some don't. it makes sense to force some settings to the whole plate, but not the whole project. 

**Describe the solution you'd like**
allow users to change settings based on the plate. Simply allow changes to the settings under the "objects" tab when a plate is selected.

**Describe alternatives you've considered**
have to change the settings back and forth or use a new project file

Hologos, 2023-07-24:
I really miss this feature as well.

QingZhangBambu, 2023-08-02:
We will investigate that.

Morialkar, 2023-10-24:
I would also appreciate this, when prepping multiple plates to start them back to back, I sometimes need one plate in Per Object printing mode and some plates in Per Layer. Ideally I think settings that cannot be set per object at least should be available to be set per plate

mzient, 2023-11-03:
Bumping up the issue! There's no reason for anything that can be customized globally not to be customizable on a per-plate basis.

GuzziRaz, 2023-11-06:
I recently even wanted one of four plates to use a 0.2 nozzle while the rest was 0.4. The only way to accomplish this right now is to put that plate in a separate file, or changing it right before printing (easy to forget, and easy to forget changing it back when printing the other plates).

closed 2024-03-26 by SaltWei

Hologos, 2024-03-26:
@SaltWei Can we get more per plate settings? Should we open new feature request? This issue talks about more settings other than just spiral mode.

SaltWei, 2024-03-26:
> I recently even wanted one of four plates to use a 0.2 nozzle while the rest was 0.4

This is hard to be supported at the moment.
Nozzle diameter is printer profile. And can not use two different nozzle diameters in one 3mf file.