bambulab/BambuStudio #2081: I would like a docker container that runs a web app version of bambu studio

kriordan37, 2023-07-19:
**Is your feature request related to a problem? Please describe.**
It is frustrating having bambu studio installed on multiple pcs. since the printer supports LAN mode i wish that the bambu studio also operated with a LAN mentality

**Describe the solution you'd like**
I would like a docker container that runs a web app version of bambu studio

**Describe alternatives you've considered**
a vm with bambu studio installed is not nearly as lightweight or convenient

**Additional context**
This is a big ask but would be great for business environments.

psiberfunk, 2023-07-20:
I'm not sure how reasonable this is considering how bambu slicer is fundamentally based on a codebase that's not particularly web-friendly.  You would do well to consider just using a VM via a web interface... it's unlikely you'll get any real focus on such a major platform code shift.

kylek29, 2023-07-22:
There's one for OrcaSlicer, it's conceivable you maybe able to install BambuStudio on it since there's a lot of compatibility overlap.

Just note that it is a tad cumbersome depending with how you want to run it .. if you're aiming to remote it via your phone, you'd need to make modifications to the supervisord.conf of the container -- primarily, I had to force a fixed higher resolution (these slicers don't play well with lower resolutions), allow remote connections, and force an input mode. That way I could connect to it using an Android VNC app, because NoVNC's in the browser is not too friendly with touch input on small screens. I think I uploaded those changes to my GitHub, but may have forgot.

Ideally, Bambu would implement and expand upon the PrusaSlicer CLI mode that allows for calling interacting via exterior GUI's.

0rphon, 2023-09-29:
ive been thinking about making a container with vnc for this, but a native webapp would be much better so i fully support this. that being said, i think creating a web version of their slicer would require a **_lot_** of work and only apply to an extremely niche part of the user base.