bambulab/BambuStudio #2082: Unable to Select Filament Type for AMS from Bambu Studio

theChef613, 2023-07-19:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Load filament into AMS
Go to the Device tab
edit the filament information
able to change color, but not able to select filament type

### Actual results

Able to change color but no Filament type dropdown

### Expected results

expected to be able to select filament type. able to do so from Bambu Handy App and then is updated in Bambu Studio

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12092060/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

eastabrooka, 2023-07-21:
+1 
[image]
I've tried left click, Right Click, Using tab and arrow keys. 

Won't let me select anything.

eastabrooka, 2023-07-21:
@theChef613 - Your bug stands, But I just downloaded Beta V1.7.0.65 Available from : 
https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.65

Heres the dropdown.  
[image]

If you want to get printing with generic filament, that's how I fixed my issue.

theChef613, 2023-07-21:
@eastabrooka thank you! going to switch to that version now

closed 2023-07-21 by theChef613