bambulab/BambuStudio #2085: BambuStudio doesn't seem to comprehend nested models

WayneManion, 2023-07-20:
### Bambu Studio Version

1.7.0 beta 2 (Fedora AppImage running on Arch)

### Where is the application from?

Bambu Lab github releases

### OS version

Arch

### Printer

Bambu P1P

### How to reproduce

Open a model that has a large opening on the inside. Use the space within that opening for other (smaller) models. BambuStudio throws warnings about the models not being far enough apart. 

Screenshot:
https://ibb.co/dtdsQWc

### Actual results

I don't know. If I see warnings I just abandon the nested models and print them later. 

### Expected results

Other slicers that I use with my other printers handle this scenario with no warnings or errors. Example: Cura 5.4.0 slicing for my Sovol SV06 Plus and Ender 3 Pro. 

### Project file & Debug log uploads

[NestedModels.zip](https://github.com/bambulab/BambuStudio/files/12110925/NestedModels.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-07-21:
Hi @WayneManion 

thanks for reporting this issue

this warning is just used to mention there may has some objects too close, it can be skipped which won't affect the slicing
it uses the convex hull to do the detection which is not accurate for your nested cases
we had considered to use the real polygons to do the detection before(which will have an accurate result)
however, the performance will be bad

you can just go ahead for slicing and if there are print conflicts, there will be serious warnings
[image]

lanewei120, 2024-01-16:
as explained above, not a bug, close it

closed 2024-01-16 by lanewei120