bambulab/BambuStudio #2086: Add Multiple Lines of Text

CredibleBS, 2023-07-20:
It does not appear possible to add a second line/multiple lines of text to an object in the slicer. I know that the slicer is not the ideal place to be doing modeling like this, however if they option is there, adding multiple seems logical.

You can duplicate a second text field onto the same object, but by doing so  you are unable to place it in any usable manner. But to be fair I am attempting to do it on a round object. However the first line works as expected.

I am unsure if it is appropriate to call that a bug as it does not appear to be an actual function of the application as opposed to a workaround not working entirely.

Potential solution:
An option to add multiple lines of text from within the slicer.

Alternatives attempted:
You can duplicate a second text field onto the same object, but by doing so  you are unable to place it in any usable manner. Also, doing so causes some of the text to be placed at odd angles, screenshot provided below.

Also, doing so causes some of the text to be placed at odd angles, screenshot provided.

I am unsure if it is appropriate to call that a bug as it does not appear to be an actual function of the application as opposed to a workaround not working entirely.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

Majestic7979, 2023-07-25:
Definitely something that would be great to have and I already have a business need for this to personalize some products I sell without having to load into CAD. I want this a lot. To be able to draw a wrap box and have the text automatically wrap would be even better. But multiple text with alignment options would be nice.

schemaczar, 2026-08-17:
It is ridiculous that Studio has good text features but not this.

Haidiye00, 2026-09-22:
https://github.com/bambulab/BambuStudio/releases/tag/v02.08.04.57
Hello, please update to the latest version. The latest version is already supported. Thank you for your support. If you have any further questions, please feel free to raise them. We will continue to communicate and thank you again

closed 2026-09-22 by Haidiye00

schemaczar, 2026-09-22:
> Haidiye00

Hi, thank you but you have a link to a beta release. I don't usually install beta releases. The current release version is 2.8.2. When can we expect a full release version of 2.8.4?

Haidiye00, 2026-09-23:
Hello, I am not sure about the specific release time, but based on previous releases, it is estimated to be in the fourth quarter of this year. Suggest using the latest beta version first. Windows can download a zip file for medical examination. Thank you